cesanta/mongoose #453: high CPU on raspberry pi

fuchsc, 2015-01-02:
Hi,

I embedded mongoose 5.2 into my raspberry pi application and it basically works perfect.
Unfortunately it consumes high CPU.
I found out, that the select syscall in mg_poll_server call never blocks for the defined timeout (1000 ms) even if no traffic at all is going on because server->ctl[1] always seems to receive data.
Is there anything I have to configure when running mongoose on raspberry?

thx,
Clemens

cpq, 2015-01-02:
Thanks.
What is the value of `server->ctl[1]` ?

Tip: you can add `-DNS_DISABLE_SOCKETPAIR` to your compile flags to disable socket pair. That will also make `mg_wakeup_server()` function.

cpq, 2015-03-21:
Hi @fuchsc , any update on this issue?

cpq, 2015-03-24:
No feedback, closing this.
Please reopen if still an issue.

closed 2015-03-24 by cpq