cesanta/mongoose #457: license checking.....

junyoungkim, 2015-01-12:
hi, 

Currently, I am considering to use a https://github.com/Gregwar/mongoose-cpp which is full re-implemented in cpp from original. 

AYK, Cpp implementations I mentioned is under a MIT license, but this is LGPL-V2.

If there is problems to use it, please give your words to warn me to prevent to use it. 

thanks for your considerations.

cpq, 2015-01-13:
Hi Todd,
it is remarkable to request a warning from the developer of the library :-)
But to be clear on it, in a nutshell: mongoose is under dual license: GPLv2 and commercial.
That means that if your project is not commercial (open source), then GPLv2 covers it perfectly, and the usage is free.
On the other hand, if mongoose is used commercially (e.g. in a paid product that ships to customers) then a license fee should be paid to us (Cesanta Software).

closed 2015-01-13 by cpq

junyoungkim, 2015-01-14:
@cpq  thanks for your comments ;)