cesanta/mongoose #467: buffer overflow detected

bitkevin, 2015-01-30:
### Error
```
*** buffer overflow detected ***: ./xxxxx terminated
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x73e0f)[0x7f915f140e0f]
/lib/x86_64-linux-gnu/libc.so.6(__fortify_fail+0x5c)[0x7f915f1d87dc]
/lib/x86_64-linux-gnu/libc.so.6(+0x10a6a0)[0x7f915f1d76a0]
/lib/x86_64-linux-gnu/libc.so.6(+0x10b727)[0x7f915f1d8727]
./xxxxx(ns_mgr_poll+0x3b6)[0x4b7036]
./xxxxx(mg_poll_server+0x9)[0x4bfcf9]
./xxxxx(_ZN15ShareRepository14runHttpdThreadEv+0xbf)[0x49420f]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb1bf0)[0x7f915fa60bf0]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x8182)[0x7f91602f3182]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f915f1c7fbd]
```

### Code
```
int main(void) {
  struct mg_server *server = mg_create_server(NULL, NULL);
  mg_set_option(server, "document_root", ".");
  mg_set_option(server, "listening_port", "8080");

  for (;;) {
    mg_poll_server(server, 1000);  // Infinite loop, Ctrl-C to stop
  }
  mg_destroy_server(&server);

  return 0;
}
```

I just put some static files under `document_root`.

cpq, 2015-03-04:
Thanks Kevin.
How can I reproduce that? Just compile and run, or did you put some traffic against the server?

cpq, 2015-03-24:
No feedback - closing this.

closed 2015-03-24 by cpq