cesanta/mongoose #470: Suggestions for a URI parser for implementation of a REST API

Mircho, 2015-02-10:
Hi,
I've been struggling to use a library that provides uri parsing and more importantly access to segments of the path. That is quite convenient in my situation where I am trying to create a meaningful REST API.
The library I settled on is uriparser (http://uriparser.sourceforge.net/)
The thing is that mongoose and uriparser do not play along. Mongoose will, conveniently, unescape utf charactes in an URL and convert them back to utf so conn->uri would contain a valid readable string. On the other hand uriparser would ignore non ascii characters and the stream processing would stop once a character that is not recognized is met. As uriparser claims conformity with and RFC 3986, it seems that the other approach would be to allow it access to the raw uri from the request. Right now there are no facilities in mongoose to skip the processing of the uri or to provide a raw uri in a different buffer.
I really don't know whether it makes sense at all. I am more or less laying some observations after couple of days struggling with uriparser.
Or I can put it in a different way - can someone suggest to me another lib to use that is as easy (low dependencies etc.) as mongoose to access segments of an URI in order to build a REST API?

cpq, 2015-02-10:
Thanks Mircho.
Could you come up with an "ideal" API for such hypothetical, no-dependency URI parser?

Mircho, 2015-02-10:
Well I would say that uriparser does the job well - once parsed an URI is split into it's parts. And especially for the case when developing a REST API - provides means for access of individual segments of an URI - /api/tools/батарея/ниво. 
In the case of uriparser the representation of the segments is a linked list which allows to scan all segments sequentially. It would be trivial to add a method for index based access to a segment.
As I said - I would have used the uriparser only if I had a way to access the raw url from the request and pass it to uriparser which will take care for the url encoded UTF8 symbols. If I was looking at only lating letters mongoose and uriparser would have worked perfectly together.
Thanks for a great piece of work - mongoose that is.

cpq, 2015-03-08:
If having non-decoded symbols in URI is critical, I suggest to use Fossa, https://github.com/cesanta/fossa
It's HTTP API is similar to Mongoose, see examples. Parsed HTTP message though is not URL-decoded.

cpq, 2015-03-24:
Closing this.

closed 2015-03-24 by cpq