cesanta/mongoose #471: Mingw-w64 Execution Issues

danriegsecker, 2015-02-10:
I have a simple application that I can compile in both Microsoft Visual Studio 2010 and Mingw-w64. The exact code compiled with Visual Studio 2010 functions as I would expect, but when I compile the same code with Mingw-w64 it does not function as expected. I desire to use Mingw-w64 targeting i686 32-bit Windows. I am using the version of the compiler found at... http://sourceforge.net/projects/mingw-w64/files/Toolchains%20targetting%20Win32/Personal%20Builds/mingw-builds/4.9.2/threads-win32/sjlj/

i686-4.9.2-release-win32-sjlj-rt_v3-rev1.7z

Below is the simple code...

#include <stdlib.h>
#include "../include/mongoose.h"

int main (int argc, char *argv[])
{
    struct mg_server *server = mg_create_server (NULL, NULL);
    if (!server) {
        puts ("unable to create http server");
        return EXIT_FAILURE;
    }
    else
        puts ("http server created successfully");
    const char *result = mg_set_option (server, "listening_port", "8085");
    if (result) {
        printf ("error setting http port of %s\n", result);
        return EXIT_FAILURE;
    }

    result = mg_set_option (server, "document_root", "C:\\html");
    if (result)
        printf ("error setting document root of %s\n", result);

    // Serve request. Hit Ctrl-C to terminate the program
    printf ("Starting on port %s\n", mg_get_option (server, "listening_port"));
    printf ("Serving from %s\n", mg_get_option (server, "document_root"));
    for (;;) {
        mg_poll_server (server, 100);
    }

    return EXIT_SUCCESS;
}

Below is example header output as captured with "Live HTTP Headers" in Mozilla Firefox....

http://localhost:8085/test.html

GET /test.html HTTP/1.1
Host: localhost:8085
User-Agent: Mozilla/5.0 (Windows NT 5.1; rv:35.0) Gecko/20100101 Firefox/35.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Referer: http://localhost:8085/
Connection: keep-alive
Range: bytes=88-
If-Range: "54d4f1d4.160"
Cache-Control: max-age=0

HTTP/1.1 206 Partial Content
Date: Mon, 09 Feb 2015 20:24:36 GMT
Last-Modified: Fri, 06 Feb 2015 16:54:44 GMT
Etag: "54d4f1d4.160"
Content-Type: text/html
Content-Length: 72
Connection: (null)
Accept-Ranges: bytes
keep-aliveContent-Range: bytes 88-0/159
----------------------------------------------------------
http://localhost:8085/test.html

GET /test.html HTTP/1.1
Host: localhost:8085
User-Agent: Mozilla/5.0 (Windows NT 5.1; rv:35.0) Gecko/20100101 Firefox/35.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Referer: http://localhost:8085/
Connection: keep-alive

HTTP/1.1 200 OK
Date: Mon, 09 Feb 2015 20:24:40 GMT
Last-Modified: Fri, 06 Feb 2015 16:54:44 GMT
Etag: "54d4f1d4.160"
Content-Type: text/html
Content-Length: 160
Connection: (null)
Accept-Ranges: bytes
----------------------------------------------------------
Notice the mess-ups after Content-Type that are returned from mongoose resulting in Mozilla Firefox returning a "Corrupted Content Error".

spuschhof, 2015-02-11:
Hello

This looks  bit like a problem, I encountered lately.
https://groups.google.com/forum/#!topic/mongoose-users/icVpaqzrQ3Q
Maybe your error has a similar reason.

cpq, 2015-02-11:
Thanks Sebastian.
Does that mean if https://github.com/cesanta/mongoose/commit/5e3e6a6dcee39b172dbce3e94e2998f5874accb8 is reverted, MinGW build will be OK?

spuschhof, 2015-02-11:
MinGW builds work fine, unless you use a Windows Version <= Server 2003 (I don't know, if the same issues occure on Windows XP), so I don't know, if it is the same issue with MinGW-w64. I don't have a w64 Installation at hand, but it should be worth a try to use I64d. 

I think the problem might be, that you have an int64_t as the argument for content-length. So using I64d should solve the problem, because it expects a 64 bit signed int. lld expects a signed long long integer, that might be greater or smaller than 64 bit.

So yes, reverting the change should fix MinGW builds for old Windows versions. But I cannot say if it will also fix the MinGW-w64 builds.

danriegsecker, 2015-02-11:
My trials have shown that this problem occurs for me on Windows XP Pro 32-bit only. I have tested with Mingw-w64 and Win-Builds which can both target 32-bit Windows. The same code works find on Windows 7 32-bit, but fails on Windows XP Pro 32-bit.

cpq, 2015-02-17:
@mmicko , apparently your commit https://github.com/cesanta/mongoose/commit/5e3e6a6dcee39b172dbce3e94e2998f5874accb8 should be reverted, since it creates problems described in this thread.
Could you elaborate what use case that commit solve in the first place, please?

mmicko, 2015-02-17:
> @mmicko <https://github.com/mmicko> , apparently your commit 5e3e6a6

Ok, think I have found what problem is.

Now changes from yesterday make sense :)

https://github.com/cesanta/mongoose/pull/473

This should now work on all cases

Micko

mmicko, 2015-02-17:
Seams like #define _XOPEN_SOURCE 600 changes behaviour of sscanf

cpq, 2015-02-17:
Merged #473 , thank you @mmicko !

closed 2015-02-17 by cpq