cesanta/mongoose #476: Why close_local_endpoint() clear connection_param?

ellusak, 2015-02-27:
I want to do cleanup tasks when the connection is closed, but close_local_endpoint() will
clear conn->connection_param. 

I think mongoose should not touch <connection_param>, this should be cleared by 
user, not by mongoose. Now, these two situations cause different behaviors:
  * if the connection is closed because handler of MG_REQUEST return MG_TRUE, 
    connection_param is cleared
 * if the connection is closed by client, connection_param is not cleared

cpq, 2015-03-08:
My understanding is that task cleanup should be done on MG_CLOSE event.

Currently mongoose sends that event before calling close_local_endpoint():

    case NS_CLOSE:
        ...
        call_user(conn, MG_CLOSE);
        close_local_endpoint(conn);
        conn->ns_conn = NULL;
        NS_FREE(conn);
      }
      break;

Could you elaborate on what exactly happens, why can't you do a proper cleanup.

cpq, 2015-03-24:
No feedback - closing this.
Please reopen if still an issue.

closed 2015-03-24 by cpq

ellusak, 2015-03-24:
Sorry, I missed your comment.

The problem is that: close_local_endpoint() is called in several position, but call_user(conn, MG_CLOSE) is not called every time. 

If the connection is closed by client, connection_param is not cleared, mongoose will call
close_local_endpoint() without calling call_user(conn, MG_CLOSE).