cesanta/mongoose #477: No reply sent after returning MG_TRUE

Vadius, 2015-03-03:
Hello.

I faced with strange behavior of API. I'm handling (RESTful) HTTP POST request and after processing trying to reply that resource is available on some URI by sending reply:
201 Created
Location: /path/to/resource/1234

My code is:

	mg_send_header(conn, "Location", "/path/to/resource/1234");
	mg_send_status(conn, 201);
	return MG_TRUE;

In this case no reply sent to client. But if we add fake body, all become ok:

	mg_send_header(conn, "Location", "/path/to/resource/1234");
	mg_send_status(conn, 201);
	mg_send_data(conn, nullptr, 0);	// workaround to enable reply sending
	return MG_TRUE;

I suppose it's not right behavior. Or I missed something in docs.

cpq, 2015-03-08:
Change call order `mg_send_status(conn, 201);` should go first:

    mg_send_status(conn, 201);
    mg_send_header(conn, "Location", "/path/to/resource/1234");
    return MG_TRUE;

In HTTP, first goes status, then headers, then payload.
Please re-open if that does not help.

closed 2015-03-08 by cpq

Vadius, 2015-03-09:
Hello.
I tried to do as you said and watched interface with wireshark. Following code
	mg_send_status(conn, 201);
	mg_send_header(conn, "Location", "/foo/bar");

produced incomplete reply:
    HTTP/1.1 201 Created
    Location: /foo/bar
    <CRLF>

No 'TransferEncoding' header, no final chunk, no double CRLF after packet head.

cpq, 2015-03-09:
I assume that's cause you didn't send any payload. Try to send one:

    mg_send_status(conn, 201);
    mg_send_header(conn, "Location", "/foo/bar");
    mg_send_data(conn, "", 0)

Vadius, 2015-03-09:
I tried it and it causes error #479.

cpq, 2015-03-09:
On the server side I did that:

    case MG_REQUEST:
      mg_send_status(conn, 201);
      mg_send_header(conn, "Location", "/path/to/resource/1234");
      mg_send_data(conn, "", 0);
      return MG_TRUE;

On the client side I see this:

    $ curl --raw -i 127.0.0.1:8080
    HTTP/1.1 201 Created
    Location: /path/to/resource/1234
    Transfer-Encoding: chunked
    
    0
    
    0

Vadius, 2015-03-09:
As [RFC 2616](https://www.ietf.org/rfc/rfc2616.txt), pt. 3.6.1 says,

    0<CRLF>
definitely is the last chunk of body. So, HTTPClient on first zero-len chunk stops receiving response body. But then mongoose sends additional data for 'closed' response. Because almost all HTTP/1.1 clients use TCP buffer, on second response (for next request) on top of buffer will be something like

    <CRLF>0<CRLF>HTTP/1.1 200 Ok

And it causes errors on next response processing.

cpq, 2015-03-09:
Okay, got it, let me clarify: mongoose sends extra 0-length chunk at the end of response, which causes next pipelined request to fail.
Is that statement correct ?

Vadius, 2015-03-09:
Exactly.
Sorry for my poor English.

cpq, 2015-03-09:
Please verify that the commit above fixes the issue.

Vadius, 2015-03-09:
Now works perfect, thank you.

commit a91438ac (2015-03-09): Fix #477