cesanta/mongoose #496: Failed to upload multiple files with HTML form

hua-su, 2015-03-22:
There is a bug in examples/file_upload/file_upload.c. It will failed to read multiple files.

To fix the bug, the original code:

[image]

shoud be changed to:

[image]

cpq, 2015-03-22:
Thanks.
Mind sending a PR ?

hua-su, 2015-03-22:
I'm sorry, I'm not a git user. Could you kindly make a PR for this issue?

cpq, 2015-03-24:
I've just made a change to upload multiple files, https://github.com/cesanta/mongoose/commit/4cbd113d0117364e8cd1e5a0334a93f6ef21406a

It works as expected without your patch. Am I missing something?

closed 2015-03-24 by cpq