cesanta/mongoose #497: mongose / stunnel

typhoon71, 2015-03-26:
I was trying to use stunnel to ssl tunnel ympd (a web interface for MPD), which uses mongoose.c.
Sadly even if the tunnel works as expected ympd is still accepting non ssl connections.
is there any simple way to use ssl only? I couldn't figure out if I have to enable stuff in mongoose.c or have to mod ympd.
btw, where should I tell mongoose where the certificate is?
Any help is really appreciated, I am no coder but if it's not too hard I can try...

cpq, 2015-03-26:
You can tell mongoose to listen on localhost only: set `listening_port` to `127.0.0.1:PORT_NUMBER` Then only stunnel going to terminate SSL connection.

Mongoose can do SSL itself. Compile with `-DNS_ENABLE_SSL -lssl`, and set `listening_port` to `ssl://PORT_NUMBER:SSL_CERTIFICATE_PATH`. SSL cert must be in PEM format, and have both private key and cert concatenated together in a single file.

closed 2015-03-26 by cpq

typhoon71, 2015-03-26:
Thanks for the tip: setting listening_port to 127.0.0.1:PORT_NUMBER was what I needed to do.
Since ympd accepts that parameter, I went with that and stunnel.
While I would've liked to use the builtin ssl capability of mongoose, this approach was faster.
Thanks again.