cesanta/mongoose #502: clang scan-build sees use-after-free in ns_mgr_free()

jhi, 2015-04-01:
Hi, the clang scan-build tool (static analysis: http://clang-analyzer.llvm.org/scan-build.html) sees a potential code path where a use-after-free happens.

The report is nice browsable HTML but not very nice to paste here, so I'll try to summarize.

The bad line is

mongoose.c:1280	    tmp_conn = conn->next;  "Use of memory after it is freed"

and how it thinks it could be freed is:

mongoose.c:1273      ns_mgr_poll(s, 0);

mongoose.c:1159	   ns_close_conn(conn);

and the ns_close_conn() does ns_destroy_conn() which does free().

Now it seems that ns_close_conn() does the correct patching of the linked list (available_connections), so I'm not entirely certain that scan-build isn't talking baloney.  But maybe someone more familiar with the code might see better.

cpq, 2015-04-02:
Alex, please take a look at the code and close this if it's a false positive.

alashkin, 2015-04-02:
I looked through the report and code, and I agree with @xant - it's false positive coz clang doesn't understand that mgr->active_connection is NULL in reported situation.
So, I can either add some code to shut up analyzer or we can just close this issue.

cpq, 2015-04-02:
Thanks Alex. Feel free to close. Let's not add anything in our code.

xant, 2015-04-02:
cool! I might report the issue back to clang developers.
Thanks for taking the time to look at it ;)

alashkin, 2015-04-02:
Thank _you_ for your analysis ;)

closed 2015-04-02 by alashkin