cesanta/mongoose #506: Cannot get the 'hello_world' example working on windows

apaikan, 2015-04-07:
I created a simple project  on Visual Studio 10 which consists of `hello_world.c`, `mongoose.c` and `mongoose.h`. Then compiled and built without specifying any option.

When I run the example, apparently everything is fine: 
```
$ hello_world.exe
Starting on port 8080
```

I can check the port 8080 is open using telnet.  But the hello world example does not reply to any http request (e.g., `GET / HTTP/1.1`). The same problem when opening the `http://127.0.0.1:8080` from the browser: the browser waits and keeps on refreshing. 

I have been using mongoose on Linux without any problem. 

Any idea? 

P.S. Debugging into the `mongoose.c` shows that non of the `ns_call`, `ns_add_to_set` and `ns_add_to_set` is called and regardless of how many connections have been established, the `for (conn = mgr->active_connections; conn != NULL; conn = tmp_conn)` loops only once. 

```C
time_t ns_mgr_poll(struct ns_mgr *mgr, int milli) {
  struct ns_connection *conn, *tmp_conn;
  struct timeval tv;
  fd_set read_set, write_set;
  sock_t max_fd = INVALID_SOCKET;
  time_t current_time = time(NULL);

  FD_ZERO(&read_set);
  FD_ZERO(&write_set);
  ns_add_to_set(mgr->ctl[1], &read_set, &max_fd);

  for (conn = mgr->active_connections; conn != NULL; conn = tmp_conn) {
    tmp_conn = conn->next;
    if (!(conn->flags & (NSF_LISTENING | NSF_CONNECTING))) {
      ns_call(conn, NS_POLL, &current_time);
    }
    if (conn->flags & NSF_CLOSE_IMMEDIATELY) {
      ns_close_conn(conn);
    } else {
      if (!(conn->flags & NSF_WANT_WRITE)) {
        //DBG(("%p read_set", conn));
        ns_add_to_set(conn->sock, &read_set, &max_fd);
      }
      if (((conn->flags & NSF_CONNECTING) && !(conn->flags & NSF_WANT_READ)) ||
          (conn->send_iobuf.len > 0 && !(conn->flags & NSF_CONNECTING) &&
           !(conn->flags & NSF_BUFFER_BUT_DONT_SEND))) {
        //DBG(("%p write_set", conn));
        ns_add_to_set(conn->sock, &write_set, &max_fd);
      }
    }
  }
```

mmikulicic, 2015-04-07:
Thanks. Reproduced.

mmikulicic, 2015-04-07:
Hi,

As a quick workaround define `FD_SETSIZE` to something like 1024.

I'll update this bug when I figure out how we can handle this more gracefully.

---

http://support.microsoft.com/en-us/kb/111855

```
/*
 * Select uses arrays of SOCKETs. These macros manipulate such
 * arrays. FD_SETSIZE may be defined by the user before including
 * this file, but the default here should be >= 64.
 *
 * CAVEAT IMPLEMENTOR and USER: THESE MACROS AND TYPES MUST BE
 * INCLUDED IN WINSOCK.H EXACTLY AS SHOWN HERE.
 */ 
#ifndef FD_SETSIZE
#define FD_SETSIZE      64
#endif /* FD_SETSIZE */ 
```

apaikan, 2015-04-07:
Hi Marko,
 Thanks. it is working now! 
It was so quick ;)

lawrancej, 2015-04-08:
I noticed a similar issue myself on mingw for windows. I managed to get things working by using the 5.5 release, rather than the latest in master.

mmikulicic, 2015-04-08:
@lawrancej Did this introduce the problem for you a90a5a2b6abb2baf8e3c10c8d ?

As far as I understood, windows defines a small FD_SETSIZE, but actually uses another limit internally, so prior to the check in a90a5a2b6abb2baf8e3c10c8d (e.g. mongoose 5.5) you were actually passing file descriptors bigger than the defined FD_SETSIZE, which is against the POSIX standard, but otherwise works fine on windows.

I think the most sensible thing to do is to not use FD_SETSIZE guard on windows, and possibly use the `iMaxSockets` runtime parameter returned by `WSAStartup`.

lawrancej, 2015-04-08:
Indeed! I ran a git bisect and found that 821e2b3a754 works fine, but b5b819b4a35 (and therefore a90a5a2b6abb2ba) breaks compilation.

For what it's worth, I copied the hello world example into the root folder to test things out and ran this command to compile things:

    gcc -g -o mongoose.exe mongoose.c hello_world.c -lpthread -lws2_32

On master, the program runs, and even brings up the usual firewall warnings on first time execution, but localhost does not respond to `curl` or the browser. But, on version 821e2b3a754 everything works as expected.

lawrancej, 2015-04-08:
Incidentally, I had tried older releases (namely, 5.5, aka 23141c2fef009), and everything worked fine up to and including 95055f604. 4502dd0c86973aa introduced compiler errors in mingw that got fixed before 821e2b3.

Cheers!

commit 1cd97150 (2015-04-17): Fix default windows FD_SETSIZE, closes #506

https://support.microsoft.com/en-us/kb/111855 describes
that the user is expected to set the FD_SETSIZE macro
before including winsock2.h
commit b8dfb69c (2015-04-17): Merge pull request #514 from cesanta/fix_setsize

Fix default windows FD_SETSIZE, closes #506
closed 2015-04-17 by mmikulicic