cesanta/mongoose #510: Directories not listed

monoped, 2015-04-14:
I get directory listings only if the client is in the same subnet as the server. If the server is contacted from a different subnet, the response code is 200 and I get an empty file. Is that intended? Or am I missing something? Everything else works fine. 
The server  (monoped.crabdance.com) is running on a BeagleBoneBlack and accessed via dynamic DNS. Mongoose is the latest version from Github, the configuration is:

    document_root /local/bernd/www
    run_as_user bernd
    listening_port 80
    access_log_file /local/bernd/access_log
    index_files index.lsp,index.html
    cgi_pattern **.lsp$

- Bernd

cpq, 2015-04-15:
That is unexpected.
Would you be able to get tcpdump captures on the server side for both good and bad cases, please?

monoped, 2015-04-15:
On 15.04.2015 08:08, Sergey Lyubka wrote:
> That is unexpected.
> Would you be able to get tcpdump captures on the server side for both
> good and bad cases, please?

Sure. Attached are 2 dumps created with

	tcpdump -vvA -i eth0 'tcp port 80'

If you need different options, please let me know.

- Bernd

13:37:05.645127 IP (tos 0x0, ttl 53, id 32855, offset 0, flags [DF], proto TCP (6), length 60)
    37.84.34.0.39483 > beagle.http: Flags [S], cksum 0xcfa9 (correct), seq 362560388, win 14600, options [mss 1452,sackOK,TS val 2372451283 ecr 0,nop,wscale 9], length 0
E..<.W@.5...%T"....p.;.P..;.......9............
.h.........	o.t.
13:37:05.645496 IP (tos 0x0, ttl 64, id 0, offset 0, flags [DF], proto TCP (6), length 60)
    beagle.http > 37.84.34.0.39483: Flags [S.], cksum 0x4722 (correct), seq 2152172542, ack 362560389, win 14480, options [mss 1460,sackOK,TS val 9468947 ecr 2372451283,nop,wscale 6], length 0
E..<..@.@.0P...p%T"..P.;.G....;...8.G".........
..|..h......
13:37:05.669110 IP (tos 0x0, ttl 53, id 32856, offset 0, flags [DF], proto TCP (6), length 52)
    37.84.34.0.39483 > beagle.http: Flags [.], cksum 0xae47 (correct), seq 1, ack 1, win 29, options [nop,nop,TS val 2372451308 ecr 9468947], length 0
E..4.X@.5...%T"....p.;.P..;..G.......G.....
.h....|..`..
13:37:05.669778 IP (tos 0x0, ttl 53, id 32857, offset 0, flags [DF], proto TCP (6), length 423)
    37.84.34.0.39483 > beagle.http: Flags [P.], cksum 0xc19d (correct), seq 1:372, ack 1, win 29, options [nop,nop,TS val 2372451308 ecr 9468947], length 371
E....Y@.5...%T"....p.;.P..;..G.............
.h....|.GET /img/ HTTP/1.1
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:37.0) Gecko/20100101 Firefox/37.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: de,en-US;q=0.7,en;q=0.3
Accept-Encoding: gzip, deflate
DNT: 1
X-Forwarded-For: 37.84.34.0
Host: monoped.crabdance.com
Cache-Control: max-age=43200
Connection: keep-alive

..J.
13:37:05.669980 IP (tos 0x0, ttl 64, id 63455, offset 0, flags [DF], proto TCP (6), length 52)
    beagle.http > 37.84.34.0.39483: Flags [.], cksum 0xabf1 (correct), seq 1, ack 372, win 243, options [nop,nop,TS val 9468960 ecr 2372451308], length 0
E..4..@.@.8x...p%T"..P.;.G....<............
..| .h..
13:37:05.673916 IP (tos 0x0, ttl 64, id 63456, offset 0, flags [DF], proto TCP (6), length 1205)
    beagle.http > 37.84.34.0.39483: Flags [P.], cksum 0x711f (correct), seq 1:1154, ack 372, win 243, options [nop,nop,TS val 9468962 ecr 2372451308], length 1153
E.....@.@.3....p%T"..P.;.G....<.....q......
..|".h..HTTP/1.1 200 OK
Transfer-Encoding: chunked
Content-Type: text/html; charset=utf-8
Transfer-Encoding: chunked

11A
<html><head><title>Index of /img/</title><style>th {text-align: left;}</style></head><body><h1>Index of /img/</h1><pre><table cellpadding="0"><tr><th><a href="?nd">Name</a></th><th><a href="?dd">Modified</a></th><th><a href="?sd">Size</a></th></tr><tr><td colspan="3"><hr></td></tr>
72
<tr><td><a href="wolfram/">wolfram/</a></td><td>&nbsp;12-Apr-2014 10:49</td><td>&nbsp;&nbsp;[DIRECTORY]</td></tr>

72
<tr><td><a href="favicon.png">favicon.png</a></td><td>&nbsp;25-Dec-2014 17:32</td><td>&nbsp;&nbsp;12.5k</td></tr>

75
<tr><td><a href="mongoose.ico">mongoose.ico</a></td><td>&nbsp;11-Apr-2014 08:45</td><td>&nbsp;&nbsp;287.9k</td></tr>

74
<tr><td><a href="mongoose.png">mongoose.png</a></td><td>&nbsp;04-Apr-2014 16:51</td><td>&nbsp;&nbsp;12.5k</td></tr>

7D
<tr><td><a href="purzelkoffer.jpg">purzelkoffer.jpg</a></td><td>&nbsp;07-Apr-2014 08:27</td><td>&nbsp;&nbsp;319.2k</td></tr>

7A
[... 84 lines cut ...]

7D
<tr><td><a href="purzelkoffer.jpg">purzelkoffer.jpg</a></td><td>&nbsp;07-Apr-2014 08:27</td><td>&nbsp;&nbsp;319.2k</td></tr>

7A
<tr><td><a href="purzelrolli.jpg">purzelrolli.jpg</a></td><td>&nbsp;07-Apr-2014 08:27</td><td>&nbsp;&nbsp;32.0k</td></tr>

0

13:34:03.915379 IP (tos 0x0, ttl 64, id 13709, offset 0, flags [DF], proto TCP (6), length 52)
    black.43972 > beagle.http: Flags [.], cksum 0xb4aa (correct), seq 324, ack 1154, win 251, options [nop,nop,TS val 4294956420 ecr 9375901], length 0
E..45.@.@......d...p...P.1.;+Z.e...........
.........H..
13:34:03.917496 IP (tos 0x0, ttl 64, id 44161, offset 0, flags [DF], proto TCP (6), length 52)
    beagle.http > black.43972: Flags [F.], cksum 0xb4af (correct), seq 1154, ack 324, win 243, options [nop,nop,TS val 9375903 ecr 4294956420], length 0
E..4..@.@......p...d.P..+Z.e.1.;...........
........
13:34:03.917710 IP (tos 0x0, ttl 64, id 13710, offset 0, flags [DF], proto TCP (6), length 52)
    black.43972 > beagle.http: Flags [F.], cksum 0xb4a4 (correct), seq 324, ack 1155, win 251, options [nop,nop,TS val 4294956422 ecr 9375903], length 0
E..45.@.@......d...p...P.1.;+Z.f...........
........z.C>
13:34:03.917890 IP (tos 0x0, ttl 64, id 44162, offset 0, flags [DF], proto TCP (6), length 52)
    beagle.http > black.43972: Flags [.], cksum 0xb4ac (correct), seq 1155, ack 325, win 243, options [nop,nop,TS val 9375903 ecr 4294956422], length 0
E..4..@.@......p...d.P..+Z.f.1.<...........
........

monoped, 2015-05-01:
Any insights?
I compiled the web server with debug enabled and tried both cases. The last lines in the good case (same subnet) were:

    close_local_endpoint 0x27a40 0 0 3145728
    ns_write_to_socket   0x279e0 1 -> 1153 bytes
    ns_close_conn        0x279e0 1
    mg_ev_handler        0x27a40 0x279e0 0 closing
    close_local_endpoint 0x27a40 0 0 1

and in the bad case (different subnet):

    close_local_endpoint 0x27a40 0 0 3145728
    ns_write_to_socket   0x279e0 1 -> 1153 bytes
    ns_close_conn        0x279e0 1
    mg_ev_handler        0x27a40 0x279e0 0 closing
    close_local_endpoint 0x27a40 0 0 1
    ns_add_sock          0x279e0 6
    accept_conn          0x279e0 6 (nil) (nil)
    ns_close_conn        0x279e0 16
    mg_ev_handler        0x27a40 0x279e0 0 closing    
    close_local_endpoint 0x27a40 0 0 16

So there are 5 additional lines here. Does that show anything to you? I can insert more debug outputs if necessary.

monoped, 2015-05-02:
Sorry, unintentionally closed it...

cpq, 2015-05-05:
Yes please enable the option

    hexdump_file <path_to_the_file>

And send the output for both good and bad case.
Thanks!

monoped, 2015-05-07:
On 05.05.2015 15:37, Sergey Lyubka wrote:
> Yes please enable the option
>
> |hexdump_file <path_to_the_file>
> |
>
> And send the output for both good and bad case.

Sergey,
here they are. It seems that in the bad case mongoose _does_ send the 
directory content, but then something happens that causes an empty 
message to be sent to the client. Strange...

Cheers, Bernd

1430995777 (nil) 192.168.2.112:80 <A 192.168.2.100:52294 0
1430995777 0x27a70 192.168.2.112:80 <- 192.168.2.100:52294 323
0000  47 45 54 20 2f 69 6d 67 2f 20 48 54 54 50 2f 31  GET /img/ HTTP/1
0010  2e 31 0d 0a 48 6f 73 74 3a 20 62 65 61 67 6c 65  .1..Host: beagle
0020  0d 0a 55 73 65 72 2d 41 67 65 6e 74 3a 20 4d 6f  ..User-Agent: Mo
0030  7a 69 6c 6c 61 2f 35 2e 30 20 28 58 31 31 3b 20  zilla/5.0 (X11; 
0040  4c 69 6e 75 78 20 78 38 36 5f 36 34 3b 20 72 76  Linux x86_64; rv
0050  3a 33 37 2e 30 29 20 47 65 63 6b 6f 2f 32 30 31  :37.0) Gecko/201
0060  30 30 31 30 31 20 46 69 72 65 66 6f 78 2f 33 37  00101 Firefox/37
0070  2e 30 0d 0a 41 63 63 65 70 74 3a 20 74 65 78 74  .0..Accept: text
0080  2f 68 74 6d 6c 2c 61 70 70 6c 69 63 61 74 69 6f  /html,applicatio
0090  6e 2f 78 68 74 6d 6c 2b 78 6d 6c 2c 61 70 70 6c  n/xhtml+xml,appl
00a0  69 63 61 74 69 6f 6e 2f 78 6d 6c 3b 71 3d 30 2e  ication/xml;q=0.
00b0  39 2c 2a 2f 2a 3b 71 3d 30 2e 38 0d 0a 41 63 63  9,*/*;q=0.8..Acc
00c0  65 70 74 2d 4c 61 6e 67 75 61 67 65 3a 20 64 65  ept-Language: de
00d0  2c 65 6e 2d 55 53 3b 71 3d 30 2e 37 2c 65 6e 3b  ,en-US;q=0.7,en;
00e0  71 3d 30 2e 33 0d 0a 41 63 63 65 70 74 2d 45 6e  q=0.3..Accept-En
00f0  63 6f 64 69 6e 67 3a 20 67 7a 69 70 2c 20 64 65  coding: gzip, de
0100  66 6c 61 74 65 0d 0a 44 4e 54 3a 20 31 0d 0a 43  flate..DNT: 1..C
0110  6f 6e 6e 65 63 74 69 6f 6e 3a 20 6b 65 65 70 2d  onnection: keep-
0120  61 6c 69 76 65 0d 0a 43 61 63 68 65 2d 43 6f 6e  alive..Cache-Con
0130  74 72 6f 6c 3a 20 6d 61 78 2d 61 67 65 3d 30 0d  trol: max-age=0.
0140  0a 0d 0a                                         ...

1430995777 0x27a70 192.168.2.112:80 -> 192.168.2.100:52294 1153
0000  48 54 54 50 2f 31 2e 31 20 32 30 30 20 4f 4b 0d  HTTP/1.1 200 OK.
0010  0a 54 72 61 6e 73 66 65 72 2d 45 6e 63 6f 64 69  .Transfer-Encodi
0020  6e 67 3a 20 63 68 75 6e 6b 65 64 0d 0a 43 6f 6e  ng: chunked..Con
0030  74 65 6e 74 2d 54 79 70 65 3a 20 74 65 78 74 2f  tent-Type: text/
0040  68 74 6d 6c 3b 20 63 68 61 72 73 65 74 3d 75 74  html; charset=ut
0050  66 2d 38 0d 0a 54 72 61 6e 73 66 65 72 2d 45 6e  f-8..Transfer-En
0060  63 6f 64 69 6e 67 3a 20 63 68 75 6e 6b 65 64 0d  coding: chunked.
0070  0a 0d 0a 31 31 41 0d 0a 3c 68 74 6d 6c 3e 3c 68  ...11A..<html><h
0080  65 61 64 3e 3c 74 69 74 6c 65 3e 49 6e 64 65 78  ead><title>Index
0090  20 6f 66 20 2f 69 6d 67 2f 3c 2f 74 69 74 6c 65   of /img/</title
00a0  3e 3c 73 74 79 6c 65 3e 74 68 20 7b 74 65 78 74  ><style>th {text
00b0  2d 61 6c 69 67 6e 3a 20 6c 65 66 74 3b 7d 3c 2f  -align: left;}</
00c0  73 74 79 6c 65 3e 3c 2f 68 65 61 64 3e 3c 62 6f  style></head><bo
00d0  64 79 3e 3c 68 31 3e 49 6e 64 65 78 20 6f 66 20  dy><h1>Index of 
00e0  2f 69 6d 67 2f 3c 2f 68 31 3e 3c 70 72 65 3e 3c  /img/</h1><pre><
00f0  74 61 62 6c 65 20 63 65 6c 6c 70 61 64 64 69 6e  table cellpaddin
0100  67 3d 22 30 22 3e 3c 74 72 3e 3c 74 68 3e 3c 61  g="0"><tr><th><a
0110  20 68 72 65 66 3d 22 3f 6e 64 22 3e 4e 61 6d 65   href="?nd">Name
0120  3c 2f 61 3e 3c 2f 74 68 3e 3c 74 68 3e 3c 61 20  </a></th><th><a 
0130  68 72 65 66 3d 22 3f 64 64 22 3e 4d 6f 64 69 66  href="?dd">Modif
0140  69 65 64 3c 2f 61 3e 3c 2f 74 68 3e 3c 74 68 3e  ied</a></th><th>
0150  3c 61 20 68 72 65 66 3d 22 3f 73 64 22 3e 53 69  <a href="?sd">Si
0160  7a 65 3c 2f 61 3e 3c 2f 74 68 3e 3c 2f 74 72 3e  ze</a></th></tr>
0170  3c 74 72 3e 3c 74 64 20 63 6f 6c 73 70 61 6e 3d  <tr><td colspan=
0180  22 33 22 3e 3c 68 72 3e 3c 2f 74 64 3e 3c 2f 74  "3"><hr></td></t
0190  72 3e 0d 0a 37 32 0d 0a 3c 74 72 3e 3c 74 64 3e  r>..72..<tr><td>
01a0  3c 61 20 68 72 65 66 3d 22 77 6f 6c 66 72 61 6d  <a href="wolfram
01b0  2f 22 3e 77 6f 6c 66 72 61 6d 2f 3c 2f 61 3e 3c  /">wolfram/</a><
01c0  2f 74 64 3e 3c 74 64 3e 26 6e 62 73 70 3b 31 32  /td><td>&nbsp;12
01d0  2d 41 70 72 2d 32 30 31 34 20 31 30 3a 34 39 3c  -Apr-2014 10:49<
[... 158 lines cut ...]
0320  22 3e 6d 6f 6e 67 6f 6f 73 65 2e 70 6e 67 3c 2f  ">mongoose.png</
0330  61 3e 3c 2f 74 64 3e 3c 74 64 3e 26 6e 62 73 70  a></td><td>&nbsp
0340  3b 30 34 2d 41 70 72 2d 32 30 31 34 20 31 36 3a  ;04-Apr-2014 16:
0350  35 31 3c 2f 74 64 3e 3c 74 64 3e 26 6e 62 73 70  51</td><td>&nbsp
0360  3b 26 6e 62 73 70 3b 31 32 2e 35 6b 3c 2f 74 64  ;&nbsp;12.5k</td
0370  3e 3c 2f 74 72 3e 0a 0d 0a 37 44 0d 0a 3c 74 72  ></tr>...7D..<tr
0380  3e 3c 74 64 3e 3c 61 20 68 72 65 66 3d 22 70 75  ><td><a href="pu
0390  72 7a 65 6c 6b 6f 66 66 65 72 2e 6a 70 67 22 3e  rzelkoffer.jpg">
03a0  70 75 72 7a 65 6c 6b 6f 66 66 65 72 2e 6a 70 67  purzelkoffer.jpg
03b0  3c 2f 61 3e 3c 2f 74 64 3e 3c 74 64 3e 26 6e 62  </a></td><td>&nb
03c0  73 70 3b 30 37 2d 41 70 72 2d 32 30 31 34 20 30  sp;07-Apr-2014 0
03d0  38 3a 32 37 3c 2f 74 64 3e 3c 74 64 3e 26 6e 62  8:27</td><td>&nb
03e0  73 70 3b 26 6e 62 73 70 3b 33 31 39 2e 32 6b 3c  sp;&nbsp;319.2k<
03f0  2f 74 64 3e 3c 2f 74 72 3e 0a 0d 0a 37 41 0d 0a  /td></tr>...7A..
0400  3c 74 72 3e 3c 74 64 3e 3c 61 20 68 72 65 66 3d  <tr><td><a href=
0410  22 70 75 72 7a 65 6c 72 6f 6c 6c 69 2e 6a 70 67  "purzelrolli.jpg
0420  22 3e 70 75 72 7a 65 6c 72 6f 6c 6c 69 2e 6a 70  ">purzelrolli.jp
0430  67 3c 2f 61 3e 3c 2f 74 64 3e 3c 74 64 3e 26 6e  g</a></td><td>&n
0440  62 73 70 3b 30 37 2d 41 70 72 2d 32 30 31 34 20  bsp;07-Apr-2014 
0450  30 38 3a 32 37 3c 2f 74 64 3e 3c 74 64 3e 26 6e  08:27</td><td>&n
0460  62 73 70 3b 26 6e 62 73 70 3b 33 32 2e 30 6b 3c  bsp;&nbsp;32.0k<
0470  2f 74 64 3e 3c 2f 74 72 3e 0a 0d 0a 30 0d 0a 0d  /td></tr>...0...
0480  0a                                               .

1430996485 0x27fb8 192.168.2.112:80 XX 37.81.47.200:15386 0

cpq, 2016-04-01:
I think this is fixed, please reopen if not.

closed 2016-04-01 by cpq