cesanta/mongoose #512: WebDAV useless

javalama, 2015-04-14:
WebDAV needs LOCK/UNLOCK support to be useful from most program, like office, OS X finder, etc.

Otherwise you'll end up with a file on the server on each save.

Who can I pay to get this implemented ASAP? :D

cpq, 2015-04-14:
LOCK/UNLOCK was not implemented because of the complexity some native clients have (mac finder, windows explorer, etc).
But to answer a question about the implementation - sure, please send a message to support@cesanta.com with your requirements.

closed 2015-04-14 by cpq

javalama, 2015-04-14:
Well, the requirements is simply to make clients work, including mac finder. I have no idea what that entails.

cpq, 2015-04-15:
Understood. Please send a list of the clients that are required to work to support@cesanta.com