cesanta/mongoose #513: Additional socket descriptor checks break windows versions

cry-inc, 2015-04-15:
The additional socket checks introduced by p_himanshu in commit b5b819b4a351b03b5dd91869582f855e124b9688 and 7da8f273d2cc0bf5ac20e1abe0ca7f5ff86996c1 break Mongoose for me. On my platform FD_SETSIZE is 64 and not a single socket file descriptor was smaller than 64. Therefore no incoming request gets through.

Environment: MSVC 2012, Windows 7, building a 64 bit application.

mmikulicic, 2015-04-15:
Yeah, we'll fix that ASAP.

In the meantime just set the preprocessor variable FD_SETSIZE to 32767 during your build.

We'll detect the actual limit via `WSAStartup`.

See
https://support.microsoft.com/en-us/kb/111855

mmikulicic, 2015-04-15:
duplicate of #506

closed 2015-04-15 by mmikulicic