cesanta/mongoose #515: MinGW hello world example issues

JustMeDaFaq, 2015-04-17:
Hey,
when trying to compile the hello_world example with MinGW im gettin errors. Ive copied both, mongoose.c and mongoose.h into the folder and used MinGW32-make command prompt. Heres the log:

Das System hat keinen Meldungstext für die Meldungsnummer 0x2350 in der Meldungs
datei Application gefunden.

(c) 2013 Microsoft Corporation. Alle Rechte vorbehalten.
Für diesen Befehl ist nicht genügend Speicher verfügbar.

C:\mongoose-master\examples>cd hello_world

C:\mongoose-master\examples\hello_world>mingw32-make -o lol.exe
cc -o hello_world hello_world.c ../../mongoose.c -W -Wall -I../.. -pthread -g -O
0
../../mongoose.c:1314:0: warning: »fileno« redefiniert [enabled by default]
 #define fileno(x) _fileno(x)
 ^
In file included from ../../mongoose.c:86:0:
c:\mingw\include\stdio.h:595:0: note: dies ist die Stelle der vorherigen Definit
ion
 #define fileno(__F) ((__F)->_file)
 ^
../../mongoose.c: In function 'mmap':
../../mongoose.c:1594:25: warning: unused parameter 'addr' [-Wunused-parameter]
 static void *mmap(void *addr, int64_t len, int prot, int flags, int fd,
                         ^
../../mongoose.c:1594:48: warning: unused parameter 'prot' [-Wunused-parameter]
 static void *mmap(void *addr, int64_t len, int prot, int flags, int fd,
                                                ^
../../mongoose.c:1594:58: warning: unused parameter 'flags' [-Wunused-parameter]

 static void *mmap(void *addr, int64_t len, int prot, int flags, int fd,
                                                          ^
../../mongoose.c:1595:23: warning: unused parameter 'offset' [-Wunused-parameter
]
                   int offset) {
                       ^
../../mongoose.c: In function 'mg_munmap':
../../mongoose.c:1646:32: warning: unused parameter 'size' [-Wunused-parameter]
 void mg_munmap(void *p, size_t size) {
                                ^
../../mongoose.c: In function 'start_process':
../../mongoose.c:2045:64: warning: unused parameter 'envp' [-Wunused-parameter]
                                   const char *env, const char *envp[],
                                                                ^
../../mongoose.c: In function 'mg_url_decode':
../../mongoose.c:2483:27: warning: signed and unsigned type in conditional expre
ssion [-Wsign-compare]
   return i >= src_len ? j : -1;
                           ^
../../mongoose.c: In function 'convert_uri_to_file_name':
../../mongoose.c:2689:56: warning: signed and unsigned type in conditional expre
ssion [-Wsign-compare]
     size_t domain_len = colon == NULL ? strlen(domain) : colon - domain;
                                                        ^
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `ns_out':
C:\mongoose-master\examples\hello_world/../../mongoose.c:404: undefined referenc
e to `sendto@24'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `ns_destroy_conn':
C:\mongoose-master\examples\hello_world/../../mongoose.c:545: undefined referenc
e to `closesocket@4'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `ns_set_non_blocking_m
ode':
C:\mongoose-master\examples\hello_world/../../mongoose.c:577: undefined referenc
e to `ioctlsocket@12'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `ns_socketpair2':
C:\mongoose-master\examples\hello_world/../../mongoose.c:595: undefined referenc
e to `htons@4'
C:\mongoose-master\examples\hello_world/../../mongoose.c:596: undefined referenc
e to `htonl@4'
[... 138 lines cut ...]
ce to `closesocket@4'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `deliver_websocket_fra
me':
C:\mongoose-master\examples\hello_world/../../mongoose.c:3019: undefined referen
ce to `htonl@4'
C:\mongoose-master\examples\hello_world/../../mongoose.c:3019: undefined referen
ce to `htonl@4'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `mg_websocket_write':
C:\mongoose-master\examples\hello_world/../../mongoose.c:3076: undefined referen
ce to `htons@4'
C:\mongoose-master\examples\hello_world/../../mongoose.c:3081: undefined referen
ce to `htonl@4'
C:\mongoose-master\examples\hello_world/../../mongoose.c:3082: undefined referen
ce to `htonl@4'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `mg_forward':
C:\mongoose-master\examples\hello_world/../../mongoose.c:4499: undefined referen
ce to `send@16'
C:\Users\JustM\AppData\Local\Temp\ccqjy5wS.o: In function `on_accept':
C:\mongoose-master\examples\hello_world/../../mongoose.c:5244: undefined referen
ce to `ntohl@4'
collect2.exe: error: ld returned 1 exit status
Makefile:14: recipe for target 'hello_world' failed
mingw32-make: *** [hello_world] Error 1

C:\mongoose-master\examples\hello_world>

JustMeDaFaq, 2015-04-17:
Ok sorry, looks like my MinGW installation was messed up, works now after reinstallng everything.

closed 2015-04-17 by JustMeDaFaq

todbot, 2017-02-08:
I'm having this same sort of issue.  What did you do to "fix" your mingw install? I've installed fresh mingw a few times now, still the same problem.

cpq, 2017-02-08:
Just to be straight, we don't use mingw build in our automated test.
We use an MSVC compiler. So these sort of issues are expected.