cesanta/mongoose #519: mongoose does not work in client mode

leonacwa, 2015-05-04:
When i use mongoose as HTTP client, i found some problems.    

HTTP response is different from HTTP request, but `process_response` process HTTP response as HTTP request through `try_parse` function.    

mongoose received all the data then call user_handler with MG_REPLY, i use mongoose to download a big video file, it received all data and save int the memory, it is unsuited.    

mongoose doesn't support speed limit, send & received buffer control, these are necessary for my project.    

PS: my English is poor, maybe what write is rude or is poor grammar , please forgive me.

cpq, 2015-05-05:
I suggest using Fossa (https://github.com/cesanta/fossa) for that. It's a superset of Mongoose, and uses very similar API and architecture.
You can catch `NS_RECV` event and store HTTP data in the file, then after the download, you can post-process the file. You can implement speed limit yourself, and you can do buffer control easily cause Fossa has all the internals opened to the user.

closed 2015-05-05 by cpq

leonacwa, 2015-05-15:
Thank you, i will try it.