cesanta/mongoose #521: SSL build

svnpenn, 2015-05-12:
I see that #481 allows now for building a web server with SSL. However
[Mongoose Free Edition][1] does not have SSL support. Is this intentional? If so
I think it should be noted on that page.

At any rate I have used [this script][2] to create a
[Windows build with SSL][3].

[1]:http://cesanta.com/mongoose.shtml
[2]:http://github.com/svnpenn/a/blob/4f014c2f18c7ad8168a039e17f/misc/mongoose.sh
[3]:http://bitbucket.org/svnpenn/a/downloads/mongoose-ssl-5.6.zip

cpq, 2015-05-13:
Hi Steven,
Yes it is intentional to have Mongoose free edition without SSL.
However anyone can build an executable with SSL  on their own, just like you did.

closed 2015-05-13 by cpq

svnpenn, 2015-05-13:
@cpq you failed to comment or take action on the fact that it is not mentioned on the download page