cesanta/mongoose #526: I use mongoose down load a file. the file is not same to the server's.

ListerTheTormentor, 2015-05-16:
I use mongoose download a file. but the download file is not same to the server's.Is there something wrong?

#include <signal.h>
#include <stdio.h>
#include <stdlib.h>

#include "mongoose.h"

static int s_received_signal = 0;
static struct mg_server *s_server = NULL;

static char s_remote_addr[64] = "127.0.0.1:8000";
static char s_remote_path[64] = "/test.zip";
static char s_local_file[64] = "/home/cll/test.zip";

static int g_time_out = 8000;
static FILE* gfp = NULL;
static int downLen = 0;

static void signal_handler(int sig_num) {
  signal(sig_num, signal_handler);
  s_received_signal = sig_num;
}

static unsigned int http_getms(void)
{
	unsigned int now;	
	struct timeval tv;
	
	gettimeofday (&tv, NULL);	

	now = tv.tv_sec * 1000 + tv.tv_usec / 1000;
	
	return now;
}

static int reply_proc(struct mg_connection* conn)
{
	int len = (int)(conn->content_len);
	int wlen;
	
	if(len > 0 /*conn->status_code == 0*/){		
		if(!gfp){
			gfp = fopen(s_local_file,"wb");			
		}
		
		if(gfp){
			wlen = fwrite(conn->content,1,len,gfp);
			printf("status = %d,write len=%d\n",conn->status_code,len);
			if(wlen != len){
				printf("write error %d,%d\n",wlen,len);
			}
			downLen += wlen;
		}
	}else{
		printf("status = %d,len = %d,content:\n%s",
			conn->status_code,len,conn->content);
	}
	
	return MG_TRUE; 

}

static int ev_handler(struct mg_connection *conn, enum mg_event ev) {
  struct mg_connection *client, *orig;

  switch (ev) {
    case MG_AUTH:
      return MG_TRUE;

    case MG_CONNECT:
      // Send request to the remote host.
      // TODO(lsm): handle connect error here.
      mg_printf(conn, "GET %s HTTP/1.0\r\nHost: %s\r\n\r\n",
                s_remote_path,
                s_remote_addr);
      return MG_TRUE;

    case MG_REPLY:
      // write replay data to file.
      reply_proc(conn);
      return MG_TRUE;
    
	case MG_CLOSE:
	  if(gfp){	  	
		fclose(gfp);
		gfp = NULL;		
	  }
	  s_received_signal = 100;
	  printf("down load end. len = %d\n",downLen);	  
	  /**/	  
	  return MG_FALSE;
	  break;
    default:
      return MG_FALSE;
  }
}

int main(void) {
	
  unsigned int preMs;    
  struct mg_connection *client;
  
  s_server = mg_create_server(NULL, ev_handler);

  mg_set_option(s_server, "listening_port", "8080");

  // Setup signal handlers
  signal(SIGTERM, signal_handler);
  signal(SIGINT, signal_handler);

  printf("Listening on port %s\n", 
  	mg_get_option(s_server, "listening_port"));

  client = mg_connect(s_server, s_remote_addr);
  if(client){
  	preMs = http_getms();
	
    while (s_received_signal == 0) {
      mg_poll_server(s_server, 1000);
	  if(http_getms() - preMs > g_time_out){
	  	break;
	  }
    }
  }
  
  mg_destroy_server(&s_server);
  
  printf("Existing on signal %d\n",s_received_signal);

  return EXIT_SUCCESS;
}

cpq, 2015-05-18:
Please hexdump both files, and show what is the difference?

rojer, 2016-05-15:
this is about mongoose 5.x, please reopen if this still applies to 6.4+

closed 2016-05-15 by rojer