cesanta/mongoose #528: content-encoding gzip support

ywhwang, 2015-05-26:
I found gzip support patch for Old-mongoose. Here is patch for new-mongoose!

//////////////
static void open_file_endpoint(struct connection *conn, const char *path,
                               file_stat_t *st, const char *extra_headers) {
  char date[64], lm[64], etag[64], range[64], headers[1000];
  const char *msg = "OK", *hdr;
  time_t t, curtime = time(NULL);
  int64_t r1, r2;
  struct vec mime_vec;
  int n;

#ifdef SERVE_GZ // = Steve start === define variables =======================
  const char *encoding;
  int	gzip_content;
#endif // SERVE_GZ = Steve end ==============================================
//  
  conn->endpoint_type = EP_FILE;
  ns_set_close_on_exec(conn->endpoint.fd);
  conn->mg_conn.status_code = 200;

  get_mime_type(conn->server, path, &mime_vec);
  conn->cl = st->st_size;
  range[0] = '\0';

#ifdef SERVE_GZ // = Steve start === check header and open file =============
  gzip_content	= 0;
  // Fetch 'Accept-Encoding' value from the header
  if((encoding=mg_get_header(&conn->mg_conn,"Accept-Encoding"))!=NULL)
  {
    if(strstr(encoding,"gzip")!=NULL)
	{
      int   path_len = strlen(path);
      char *path_gz;
	  file_stat_t mgs;
      // Append ".gz" to the path
      if((path_gz=malloc(path_len+4))!=NULL)
	  {
	    snprintf(path_gz,path_len+4,"%s.gz",path); 
		// Get the file stats to update 'Content-Length' correctly
		if(stat((const char *)path_gz, &mgs)==0)
		{
			if(conn->endpoint.fd!=-1)close(conn->endpoint.fd);
          // If file opens, adjust content length and serve it...
          if((conn->endpoint.fd=open(path_gz,O_RDONLY | O_BINARY, 0)) != -1)
		  {
			gzip_content = 1;
			conn->cl = mgs.st_size;
			// do not modify original file_stat_t st!!
			// do not modify original path
			ns_set_close_on_exec(conn->endpoint.fd);
		} }
        free(path_gz);
  } } }
#endif // SERVE_GZ = Steve end ==============================================
  // If Range: header specified, act accordingly
  r1 = r2 = 0;
  hdr = mg_get_header(&conn->mg_conn, "Range");
  if (hdr != NULL && (n = parse_range_header(hdr, &r1, &r2)) > 0 &&
      r1 >= 0 && r2 >= 0) {
    conn->mg_conn.status_code = 206;
    conn->cl = n == 2 ? (r2 > conn->cl ? conn->cl : r2) - r1 + 1: conn->cl - r1;
    mg_snprintf(range, sizeof(range), "Content-Range: bytes "
                "%" INT64_FMT "-%" INT64_FMT "/%" INT64_FMT "\r\n",
                r1, r1 + conn->cl - 1, (int64_t) st->st_size);
    msg = "Partial Content";
    lseek(conn->endpoint.fd, r1, SEEK_SET);
  }

  // Prepare Etag, Date, Last-Modified headers. Must be in UTC, according to
  // http://www.w3.org/Protocols/rfc2616/rfc2616-sec3.html#sec3.3
  gmt_time_string(date, sizeof(date), &curtime);
  t = st->st_mtime; // store in local variable for NDK compile
  gmt_time_string(lm, sizeof(lm), &t);
  construct_etag(etag, sizeof(etag), st);

#ifdef SERVE_GZ // = Steve start === modify response header =================
  if(gzip_content)
  {
  n = mg_snprintf(headers, sizeof(headers),
                  "HTTP/1.1 %d %s\r\n"
                  "Date: %s\r\n"
                  "Last-Modified: %s\r\n"
                  "Etag: %s\r\n"
                  "Content-Type: %.*s\r\n"
                  "Content-Length: %" INT64_FMT "\r\n"
                  "Content-Encoding: gzip\r\n"			// <-- Added here ...
                  "Connection: %s\r\n"
                  "Accept-Ranges: bytes\r\n"
                  "%s%s%s\r\n",
                  conn->mg_conn.status_code, msg, date, lm, etag,
                  (int) mime_vec.len, mime_vec.ptr, conn->cl,
                  suggest_connection_header(&conn->mg_conn),
                  range, extra_headers == NULL ? "" : extra_headers,
                  MONGOOSE_USE_EXTRA_HTTP_HEADERS);
  }
  else
#endif // SERVE_GZ = Steve end ==============================================
  n = mg_snprintf(headers, sizeof(headers),
                  "HTTP/1.1 %d %s\r\n"
                  "Date: %s\r\n"
                  "Last-Modified: %s\r\n"
                  "Etag: %s\r\n"
                  "Content-Type: %.*s\r\n"
                  "Content-Length: %" INT64_FMT "\r\n"
                  "Connection: %s\r\n"
                  "Accept-Ranges: bytes\r\n"
                  "%s%s%s\r\n",
                  conn->mg_conn.status_code, msg, date, lm, etag,
                  (int) mime_vec.len, mime_vec.ptr, conn->cl,
                  suggest_connection_header(&conn->mg_conn),
                  range, extra_headers == NULL ? "" : extra_headers,
                  MONGOOSE_USE_EXTRA_HTTP_HEADERS);
  ns_send(conn->ns_conn, headers, n);

  if (!strcmp(conn->mg_conn.request_method, "HEAD")) {
    conn->ns_conn->flags |= NSF_FINISHED_SENDING_DATA;
    close(conn->endpoint.fd);
    conn->endpoint_type = EP_NONE;
  }
}

miketung, 2015-07-30:
Seems this would not support cgi paths, correct?

cpq, 2015-07-31:
Could you make a pull request, and sign the CLA please
https://cesanta.com/contributors_la.shtml

rojer, 2016-05-15:
closing as stale (applies to 5.x, no PR)

closed 2016-05-15 by rojer

pfried, 2016-10-20:
This feature does improve the performance of mongoose on memory restricted devices by a factor. In our application the page loading time goes from 7s to 1,7s (About 2.5MB, compressed 500KB) if we gzip the contents and serve them with the Content-Encoding.

Also the implementation is not really hard:

- Check `Accept-Encoding: gzip` header
- Do a `stat` on the same path with an additional `.gz` suffix
- Serve the gzipped file and set the additional header

Content must be pre-compressed of course.