cesanta/mongoose #530: URL rewrites for minify(using php5-cgi)

ywhwang, 2015-05-26:
here is patch for convert_uri_to_file_name.

in mongoose.conf

url_rewrites /min/f=/min/index.php?f,/min/g=/min/index.php?g

///////////////////////////

static int convert_uri_to_file_name(struct connection *conn, char *buf,
                                    size_t buf_len, file_stat_t *st) {
  struct vec a, b;
  const char *rewrites = conn->server->config_options[URL_REWRITES];
  const char *root =
#ifndef MONGOOSE_NO_DAV
    is_dav_request(conn) && conn->server->config_options[DAV_ROOT] != NULL ?
    conn->server->config_options[DAV_ROOT] :
#endif
    conn->server->config_options[DOCUMENT_ROOT];
#ifndef MONGOOSE_NO_CGI
  const char *cgi_pat = conn->server->config_options[CGI_PATTERN];
  char *p;
#endif
  const char *uri = conn->mg_conn.uri;
  const char *domain = mg_get_header(&conn->mg_conn, "Host");
  // Important: match_len has to be declared as int, unless rewrites break.
  int match_len;size_t root_len = root == NULL ? 0 : strlen(root);

  // Perform virtual hosting rewrites
  if (rewrites != NULL && domain != NULL) {
    const char *colon = strchr(domain, ':');
    size_t domain_len = colon == NULL ? strlen(domain) : (size_t)(colon - domain);

    while ((rewrites = next_option(rewrites, &a, &b)) != NULL) {
      if (a.len > 1 && a.ptr[0] == '@' && a.len == domain_len + 1 &&
          mg_strncasecmp(a.ptr + 1, domain, domain_len) == 0) {
        root = b.ptr;
        root_len = b.len;
        break;
      }
    }
  }

  // No filesystem access
  if (root == NULL || root_len == 0) return 0;

  // Handle URL rewrites
  mg_snprintf(buf, buf_len, "%.*s%s", root_len, root, uri);
  rewrites = conn->server->config_options[URL_REWRITES];  // Re-initialize!
  while ((rewrites = next_option(rewrites, &a, &b)) != NULL) {
    if ((match_len = mg_match_prefix(a.ptr, a.len, uri)) > 0) {
      mg_snprintf(buf, buf_len, "%.*s%.*s%s", root_len, root, (int) b.len, b.ptr, uri + match_len);
      break;
    }
  }

  if (stat(buf, st) == 0) return 1;

#ifndef MONGOOSE_NO_CGI
  // Support PATH_INFO for CGI scripts.
  for (p = buf + strlen(root) + 2; *p != '\0'; p++) {
    char chOld=*p;
    
    if (*p == '/' ||*p == '?') {
      *p = '\0';
      if (mg_match_prefix(cgi_pat, strlen(cgi_pat), buf) > 0 &&
          !stat(buf, st)) {
      DBG(("!!!! [%s]", buf));
        // *p = '/';
        if(chOld=='?'){*p='\0';conn->mg_conn.query_string=p+1;}
        else *p = chOld;
        conn->path_info = mg_strdup(p);
        *p = '\0';
        return 1;
      }
      // *p = '/';
      *p = chOld;
    }
  }
#endif

  return 0;
}

ywhwang, 2015-05-26:
In prepare_cgi_environment
////////////////////
  // TODO(lsm): fix this for IPv6 case
  addenv(blk, "SERVER_PORT=%d", ri->local_port);//comment out for minify_test

ywhwang, 2015-05-26:
More corrent patch is: c->query_string must be alloced variable!!
//////////////////////////////////////////////////////in parse_http_message
    if ((ri->query_string = strchr(ri->uri, '?')) != NULL) {
      *(char *) ri->query_string++ = '\0';
#if 1	  
      ri->query_string=mg_strdup(ri->query_string);// ywhwang patch for NS_FREE&url_rewrite
#endif      
    }
//////////////////////////////////////////////////in close_local_endpoint
  // Do not memset() the whole structure, as some of the fields
  // (IP addresses & ports, server_param) must survive. Nullify the rest.
#if 1  
  if(c->query_string)NS_FREE((void*)c->query_string);// ywhwang PATCH for url-rewrite
#endif  
  c->request_method = c->uri = c->http_version = c->query_string = NULL;

/////////////////////////////////////////////convert_uri_to_file_name
static int convert_uri_to_file_name(struct connection *conn, char *buf,
                                    size_t buf_len, file_stat_t *st) {
  struct vec a, b;
  const char *rewrites = conn->server->config_options[URL_REWRITES];
  const char *root =
#ifndef MONGOOSE_NO_DAV
    is_dav_request(conn) && conn->server->config_options[DAV_ROOT] != NULL ?
    conn->server->config_options[DAV_ROOT] :
#endif
    conn->server->config_options[DOCUMENT_ROOT];
#ifndef MONGOOSE_NO_CGI
  const char *cgi_pat = conn->server->config_options[CGI_PATTERN];
  char *p;
#endif
  const char *uri = conn->mg_conn.uri;
  const char *domain = mg_get_header(&conn->mg_conn, "Host");
  // Important: match_len has to be declared as int, unless rewrites break.
  int match_len;size_t root_len = root == NULL ? 0 : strlen(root);

  // Perform virtual hosting rewrites
  if (rewrites != NULL && domain != NULL) {
    const char *colon = strchr(domain, ':');
    size_t domain_len = colon == NULL ? strlen(domain) : (size_t)(colon - domain);

    while ((rewrites = next_option(rewrites, &a, &b)) != NULL) {
      if (a.len > 1 && a.ptr[0] == '@' && a.len == domain_len + 1 &&
          mg_strncasecmp(a.ptr + 1, domain, domain_len) == 0) {
        root = b.ptr;
        root_len = b.len;
        break;
      }
    }
  }

  // No filesystem access
  if (root == NULL || root_len == 0) return 0;

  // Handle URL rewrites
  mg_snprintf(buf, buf_len, "%.*s%s", root_len, root, uri);
  rewrites = conn->server->config_options[URL_REWRITES];  // Re-initialize!
  while ((rewrites = next_option(rewrites, &a, &b)) != NULL) {
    if ((match_len = mg_match_prefix(a.ptr, a.len, uri)) > 0) {
      mg_snprintf(buf, buf_len, "%.*s%.*s%s", root_len, root, (int) b.len, b.ptr, uri + match_len);
      break;
    }
  }

#if 1	// URL rewrites can change query_string!!
  p=strchr(buf + strlen(root) + 2,'?');
  if(p&&(!conn->mg_conn.query_string||strcmp(p+1,conn->mg_conn.query_string))){
	if(conn->mg_conn.query_string)printf("before query_string='%s'\n",conn->mg_conn.query_string);
  	if(conn->mg_conn.query_string)NS_FREE((void*)conn->mg_conn.query_string);
  	conn->mg_conn.query_string=mg_strdup(p+1);
	*p= '\0';
	printf("after query_string='%s'\n",conn->mg_conn.query_string);
  }
#endif  
  if (stat(buf, st) == 0) return 1;

#ifndef MONGOOSE_NO_CGI
  // Support PATH_INFO for CGI scripts.
  for (p = buf + strlen(root) + 2; *p != '\0'; p++) {
    if (*p == '/') {
      *p = '\0';
      if (mg_match_prefix(cgi_pat, strlen(cgi_pat), buf) > 0 &&
          !stat(buf, st)) {
        DBG(("!!!! [%s]", buf));
        *p = '/';
        conn->path_info = mg_strdup(p);
        *p = '\0';
        return 1;
      }
      *p = '/';
    }
  }
#endif

  return 0;
}

rojer, 2016-05-15:
please - plain language first, _then_ code. i don't understand this.

closed 2016-05-15 by rojer