cesanta/mongoose #531: UDP receive problem.

ywhwang, 2015-05-26:
HERE is patch from fossa.
////////////////////////////////////////////////////
static void ns_handle_udp(struct ns_connection *ls) {
  struct ns_connection nc;
  char buf[NS_UDP_RECEIVE_BUFFER_SIZE];
  ssize_t n;
  socklen_t s_len = sizeof(nc.sa);

  memset(&nc, 0, sizeof(nc));
RetryReceive:// ywhwang patch for MultiReceive
  n = recvfrom(ls->sock, buf, sizeof(buf), 0, &nc.sa.sa, &s_len);
  if (n <= 0) {
    DBG(("%p recvfrom: %s", ls, strerror(errno)));
    if(n<0)errno=0; return;// ywhwang patch for MultiReceive 
  } else {
    nc.mgr = ls->mgr;
    nc.recv_iobuf.buf = buf;
    nc.recv_iobuf.len = nc.recv_iobuf.size = n;
    nc.sock = ls->sock;
    nc.callback = ls->callback;
#ifdef NS_ENABLE_COAP
    nc.handler = ls->handler;     // Protocol-specific event handler //ywhwang ADD for UDP packet
#endif
    nc.user_data = ls->user_data;
    nc.proto_data = ls->proto_data;
    nc.mgr = ls->mgr;
    nc.listener = ls;
    nc.flags = NSF_UDP;
    DBG(("%p %d bytes received", ls, n));
    ns_call(&nc, NS_RECV, &n);
    goto RetryReceive;// ywhwang patch for MultiReceive
  }
}

rojer, 2016-05-15:
Mongoose 6.x is based on Fossa. additionally, @alashkin has done a lot of work on UDP recently.
please open new issues if you still have problems with UDP in Mongoose after 6.4

closed 2016-05-15 by rojer

snahmad, 2019-05-10:
Is this issue related with udp receive not data on Windows platform.