cesanta/mongoose #534: More TCP/UDP support required

ywhwang, 2015-05-26:
Here is some patch.
//////////////////////////////////////

/// header processing for debug
const char *mg_get_header_all(const struct mg_connection *ri,char headerBuf[2048]) {
  int i;

	headerBuf[0]=0;
  for (i = 0; i < ri->num_headers; i++)
    if (ri->http_headers[i].name[0]&&ri->http_headers[i].value[0]){
      strcat(headerBuf,ri->http_headers[i].name);strcat(headerBuf,":");
	strcat(headerBuf,ri->http_headers[i].value);strcat(headerBuf,"\n");
    }

  return headerBuf;
}

#define mg_get_server_from_mgr(mgr) 	((struct mg_server *)mgr)
#define mg_get_nsmgr(server) 	((struct ns_mgr *)server)
// to support multi-bind
struct ns_connection *mg_bind(struct mg_server *server,const char* listening_port,void* user_data)
{
	struct ns_connection *c = ns_bind(mg_get_nsmgr(server), listening_port,
                                        mg_ev_handler, user_data);
	return c;//mg_get_nsmgr(server)->user_data == server_data
}

// to support TCP/UDP protocol
// to support UDP protocol
static size_t ns_outto(struct ns_connection *nc, union socket_address *p_sa, const void *buf, size_t len) {
  if (nc->flags & NSF_UDP) {
    long n = sendto(nc->sock, (const char *) buf, len, 0, &p_sa->sa,
                    sizeof(p_sa->sin));
    DBG(("%p %d send %ld (%d %s) to %s:%hu", nc, nc->sock, n, errno, strerror(errno), inet_ntoa(p_sa->sin.sin_addr), ntohs(p_sa->sin.sin_port)));
    return n < 0 ? 0 : n;
  } else {
    return 0;
  }
}
int ns_sendto(struct ns_connection *conn, union socket_address *p_sa, const void *buf, size_t len) {
  return (int) ns_outto(conn, p_sa, buf, len);
}
int ns_sendTo(struct ns_connection *conn, const char *address, const void *buf, size_t len) {
  union socket_address sa;
  int use_ssl, proto;
  char cert[100], ca_cert[100];

  ns_parse_address(address, &sa, &proto, &use_ssl, cert, ca_cert);
  return ns_sendto(conn, &sa, buf, len);
}

//////////////////////////////
int mg_is_ssl(struct mg_connection *c) { // ywhwang add for redirection!!
#ifdef NS_ENABLE_SSL
  struct connection *conn = MG_CONN_2_CONN(c);
  return conn&&conn->ns_conn&&conn->ns_conn->ssl? 1:0;
#else
  (void)c;
  return 0;
#endif
}

////////////////connection conversion
struct ns_connection *mg_get_nc_conn(struct mg_connection *conn)
{
  struct connection *c = MG_CONN_2_CONN(conn);
  return c->ns_conn;
}
void NS_close_conn(struct ns_connection *nc)
{
  ns_close_conn(nc);
}

struct mg_connection *NS_CONN_2_MG_CONN(struct ns_connection *nc) ;
int IS_ACTIVE_MG_CONN(struct ns_connection *nc) //very important function!!!
{
	if(!nc||!nc->mgr){printf("IS_ACTIVE_NS_CONN: nc%p have not ns_mgr!!\n",nc);return 0;}
	{
		struct mg_connection *c;		
		struct mg_server *s;

		s=mg_get_server_from_mgr(nc->mgr);
		if(!s){printf("IS_ACTIVE_NS_CONN: nc%p have not mg_server!!\n",nc);return 0;}//??
		// 2nd verify !!
		for (c = mg_next(s, NULL); c != NULL; c = mg_next(s, c)) {
			if(mg_get_nc_conn(c)==nc)return 1;
		}
		if(!c){/*printf("IS_ACTIVE_NS_CONN: nc%p is not a mg_conn!!\n",nc);*/return 0;}//very important CASE!!!// not a mg_conn!!!!(for mgr1,2 -> TCP connections!!!)
	}
	return 1;
}
struct mg_connection *NS_CONN_2_MG_CONN(struct ns_connection *nc) 
{
  if(!IS_ACTIVE_MG_CONN(nc))return NULL;// verify
  if (nc != NULL && nc->user_data != NULL) {
    return & ((struct connection *) nc->user_data)->mg_conn;
  } else {
    return NULL;
  }
}

wilx, 2015-09-01:
Ugh. This is GitHub. You should use a pull request to contribute.

rojer, 2016-05-15:
this no longer applies to MG after 6.0 rewrite.

closed 2016-05-15 by rojer