cesanta/mongoose #543: remove file caused 404

songtzu, 2015-06-11:
here is my code
mg_send_file(conn, filepath, NULL); 
if( remove( filepath) != 0 ){
}

when network condition is not good, browser got 404.

cpq, 2015-06-11:
mg_send_file() schedules file for sending, it does not send it immediately. You should remove the file when request is over.

closed 2015-06-11 by cpq

songtzu, 2015-06-12:
how could I know it finished sending?
any callback?
could you tell me where the function exactly send the file?

cpq, 2015-06-18:
Actually, thinking more about it..

    mg_send_file(conn, filepath, NULL); 
    if( remove( filepath) != 0 ){
    }

That should work if you're on UNIX based system.
mg_send_file() opens   a file descriptor and remove() removes a reference to a file, so a file will be sent successfully.

What OS you're running on?