cesanta/mongoose #544: sock > FD_SETSIZE on Windows

Tigzy, 2015-06-12:
Similar issue than: https://github.com/cesanta/mongoose/issues/90

Hello, I just found a similar issue where FD_SETSIZE is checked in ns_add_to_set, and when the socket created is highler to that value (surprisingly that happens to me only in release mode) then ns_mgr_poll will always return 0 because of empty read_set/write_set. Plus, not sure why but this bug also leads to high CPU usage (25% with bug, 1% without bug)

Maybe a clue:
mongoose.c, line 836:
// NOTE(lsm): on Windows, sock is always > FD_SETSIZE
mongoose.c, line 1068:
if ( (sock != INVALID_SOCKET) && (sock < FD_SETSIZE) ) {
(note, this is the check I'm talking about earlier, in ns_add_to_set)

After changing #define FD_SETSIZE definition to 2048, no more bug.
But I don't think this is the correct way to do, ugly workaround.

mmikulicic, 2015-06-13:
Hi, thanks for the bug report.

http://support.microsoft.com/en-us/kb/111855 describes how this preprocessor macro is actually intended to be redefined by the user.

Windows exposes the actual runtime limit via the `iMaxSockets` property.

My understanding is that the socket API on windows is an underdog. We might need to switch to the native api and use `WaitForMultipleObjects`.

In any case we should at least have better reporting when file descriptors get bigger than FD_SETSIZE, since it leads to confusing behaviour (i.e. increased cpu usage)

mmikulicic, 2015-06-13:
Hm, WaitForMultipleObjects can only wait for 64 objects at a time. IO completion ports might be a better alternative.

Tigzy, 2015-06-15:
"http://support.microsoft.com/en-us/kb/111855 describes how this preprocessor macro is actually intended to be redefined by the user."
Maybe it's better to move it in the header then? because I use mongoose as a .lib, so my head project cannot redefine it (precompiled with the value of 1024).

mmikulicic, 2015-06-15:
You need to recompile the mongoose binary when you change the FD_SETSIZE.
The `fd_set` data type is a fixed size buffer.

For now you have to compile your library with a FD_SETSIZE limit high enough for your application domain, and low enough that it works on the windows versions you target.

We'll be working on a `poll` based event manager, targeting windows and other unices. No ETA yet.

closed 2015-06-15 by mmikulicic