cesanta/mongoose #548: my first web server failed

longmarchw, 2015-06-17:
my web server most code:
------------------------

static int ev_handler(struct mg_connection *conn, enum mg_event ev) {
  switch (ev) {
    case MG_AUTH: return MG_TRUE;
    case MG_REQUEST:
	  mg_send_file(conn, "index.html", NULL);
      return MG_MORE;
    default: return MG_FALSE;
  }
}

main( ){
...
mg_set_option(server, "listening_port", "8080");
mg_set_option(server, "document_root", ".");
...
}

------------------------
index.html:
...
img id="flowImage" src="image/12.bmp"
...
------------------------
the windows path:

D:\debug\mywebserver.exe
D:\debug\index.html
D:\debug\image\12.bmp

---------------------------------
Run mywebserver.exe, i can see the index.html is open in IE, but the bmp picture is not showen.

what's wrong with me? Please help me.thx.

Thuenem, 2015-06-18:
I think the experienced behavior happens because you always serve index.html for each request you get. Even if the browser requests the resource image/12.bmp your ev_handler routine will return index.html. To solve this issue there are two possible solutions.

If you want to serve static files only there is no need to use the ev_handler callback - mongoose will acess the files in document_root and serve it for you.

The ev_handler callback should be used if you want to extend the functionality of mongoose. For this purpose you are able to react on certain events triggered while processing the incoming HTTP messages.

cpq, 2015-06-18:
@Thuenem is correct. The easiest way to fix your code is:

    case MG_REQUEST:  return MG_FALSE;

closed 2015-06-18 by cpq