cesanta/mongoose #549: Sporadic accept()

OlafBerens, 2015-06-23:
Hello !

We have sparodic accept() success / failures

I changed 
if ((sock != INVALID_SOCKET) && (sock < FD_SETSIZE)) 
to
if ((sock != INVALID_SOCKET) && (set->fd_count < FD_SETSIZE)) 

(i have FD_SETSIZE = 64 right now)

and now it works perfectly

please have a look and verify

Olaf Berens
PROGNOST Systems GmbH

cpq, 2015-06-24:
Thanks Olaf.
What is your operating system?

OlafBerens, 2015-06-24:
Latest patches installed !

using this setting from SDK 7.1

:

cpq, 2015-06-25:
Thanks, but what is the operating systems? Is it Windows?

mmikulicic, 2015-06-25:
@OlafBerens, Hi!

```c
if ((sock != INVALID_SOCKET) && (sock < FD_SETSIZE)) 
```

Checks that a socket number is within the bounds of the size of the fd set structure. While:

```c
if ((sock != INVALID_SOCKET) && (set->fd_count < FD_SETSIZE))
```

is always true since the fd_count field will always be within those bounds.

Microsoft describes how to increase the size of that structure in http://support.microsoft.com/en-us/kb/111855, suggesting to redefiend the FD_SETSIZE macro when building the library.

We already discussed this issue in #544, #513 and #506.
The current situation is suboptimal at least, and we'll move to a `poll` for supported windows versions.
In the meantime you should just change the FD_SETSIZE variable to a higher number (e.g. 1024).

OlafBerens, 2015-06-25:
Marko,

thanks for your reply. please find attached my log
search for   "accept_conn" and you will see the returned sock number in 
the range of 800 to 1200.

Therefore the mongoose code sometimes work, sometimes not   (1024 limit)

the "set" is a fixed length array and MUST be checked

the "sock" is a handle of any number between 1 and 65535.

see:
FD_SET(sock, set);

code:
#define FD_SET(fd, set) do { \
    u_int __i; \
    for (__i = 0; __i < ((fd_set FAR *)(set))->fd_count; __i++) { \
        if (((fd_set FAR *)(set))->fd_array[__i] == (fd)) { \
            break; \
        } \
    } \
    if (__i == ((fd_set FAR *)(set))->fd_count) { \
        if (((fd_set FAR *)(set))->fd_count < FD_SETSIZE) { \
            ((fd_set FAR *)(set))->fd_array[__i] = (fd); \
            ((fd_set FAR *)(set))->fd_count++; \
        } \
    } \
} while(0)

"sock" should never be the index of the array, "sock" is the value of the 
array-item and has nothing to do with the array index or size ?!?!?!

Correct me if i am wrong
With kind regards

Olaf Berens 

Forschung & Entwicklung
Research & Development

PROGNOST Systems GmbH
Daimlerstraße 10
48432 Rheine (Germany)

Phone:    +49-(0)5971-80819.31
Fax:         +49-(0)5971-80819.42
Email:      olaf.berens@prognost.com
Disclaimer

@OlafBerens, Hi!

mmikulicic, 2015-06-25:
according to that code it looks that it's not the index in the set; perhaps I got confused or it's implementation specific.

cpq, 2015-06-25:
I think that on windows that check should be disabled.
Olaf could you check that please?

```
if ((sock != INVALID_SOCKET) 

#ifndef _WIN32
   && (sock < FD_SETSIZE)
#endif
) 
```

davidm-mdxt, 2016-04-18:
We're suffering from this issue on Windows too. I believe the underlying issue is that Windows socket handles are not small positive integers as on Linux, and so Windows uses a different fd_set structure (and different FD_SET logic) to Linux. Your check `&& (sock < FD_SETSIZE)` does not work when socket handles are larger than FD_SETSIZE on Windows. Increasing FD_SETSIZE is a workaround, but I believe a far better fix is that in cpq's comment above - wrapping `#ifndef _WIN32` around the `&& (sock < FD_SETSIZE)`. Note also that Windows ignores the `nfds` parameter to `select()`, so `max_fd` doesn't really matter here on Windows.

cpq, 2016-04-18:
@davidm-mdxt do you have a working workaround for windows?

davidm-mdxt, 2016-04-18:
@cpq Absolutely, yes - your suggestion above is working for me on Windows (in Mongoose 5.6):
```
static void ns_add_to_set(sock_t sock, fd_set *set, sock_t *max_fd) {
  if ( (sock != INVALID_SOCKET) 
#ifndef _WIN32
       && ( sock < FD_SETSIZE )
#endif
       ) {
    FD_SET(sock, set);
    if (*max_fd == INVALID_SOCKET || sock > *max_fd) {
      *max_fd = sock;
    }
  }
}
```

davidm-mdxt, 2016-04-18:
The equivalent code in the current Mongoose 6.4 release is:

```
void mg_add_to_set(sock_t sock, fd_set *set, sock_t *max_fd) {
  if (sock != INVALID_SOCKET) {
    FD_SET(sock, set);
    if (*max_fd == INVALID_SOCKET || sock > *max_fd) {
      *max_fd = sock;
    }
  }
}
```

This fixes the issue on Windows (there's no unnecessary checking that `sock < FD_SETSIZE` there) but unfortunately removes this important check on Linux. From the documentation: "Executing FD_CLR() or FD_SET()  with a value of fd that is negative or is equal to or larger than FD_SETSIZE will result in undefined behavior." Looking at the definition of FD_SET (`#define __FD_SET(d, set)    (__FDS_BITS (set)[__FDELT (d)] |= __FDMASK (d))`) this 'undefined behaviour' is a buffer overrun. I believe this needs fixing on Linux now.

cpq, 2016-04-22:
@rojer does it make sense to integrate it ?

rojer, 2016-04-22:
i plan to do some testing on windows and keep this in mind.

rojer, 2016-05-15:
handling of sockets > FD_SETSIZE has been fixed in https://github.com/cesanta/mongoose/commit/203d4a79e282b75b7265b1fce57cb5720fe3d62f
please reopen if the issue is still there.

closed 2016-05-15 by rojer