cesanta/mongoose #555: windows memory leak with php-cgi.exe

asainnp, 2015-07-12:
every single php request cause php-cgi.exe to stay in processes list.
it is hard to find bug, because classic task-manager do not list them. but RAMMAP.exe shows them.
after few days of working my web-server caused 2-3gb memory consuption.
again: it is not viewable directly in task-manager that mongoose is guilty, but ... it is inside rammap.exe.
----
older mongoose versions that i tested have this problem in ~all windows versions, versions from last few months are better one win7 machine seems ok, but other not... Win8 100% error always.
----
so to reproduce a problem, run any mongoose on win8 and keep refreshing rammap.exe process list while opening some php page.
----
after few days of debugging, and testing:
uncommenting 2 lines inside mongoose.c SOLVED a problem.
can someone please explain why this two lines are commented:
function static process_id_t start_process( ...
...
  //CloseHandle(pi.hThread);    // <- uncommenting 2 lines
  //CloseHandle(pi.hProcess);  //       solved a problem

  return pi.hProcess;
}
why is there a comment and ... is do you think it is safe to uncomment those lines? 
... this stdin/stdout duplicating and controling running process was always a trouble in windows and behaves unpredictable sometimes. But is very important problem for mongoose while not having fast-cgi socket implementation.
it seems that handle was not closed... and win8 saves half-dead php-cgi.exe processes' memory.

exmex, 2015-10-02:
I don't know which version you're using but in my it clearly says:
```
/* TODO(lsm): check if we need close process and thread handles too */
```

And apparently this uses it (5682)
```
} else if (start_process(opts->cgi_interpreter, prog, blk.buf, blk.vars, dir, fds[1]) != 0) {
```

alashkin, 2016-09-26:
Fixed in https://github.com/cesanta/mongoose/commit/2e748fdbf129de08255ae07671d235efc466b58c

closed 2016-09-26 by alashkin