cesanta/mongoose #558: Simple forward proxy

nevson, 2015-07-23:
Hi, I want to map several services running on different ports onto one central URI e.g. `https://myserver.int` which is public accessible. Other ports will be blocked for the outside so that these other "hidden" services are only locally accessible. I.e. I want to achieve following mapping:

```
https://myserver.int/service1/ -> tcp://localhost:8081/
https://myserver.int/service2/ -> tcp://localhost:8082/
...
```
My initial thought was to write a proxy which forwards the requests to the servers depending on the URI. Is this the correct approach to achieve this or is there a better way or am I understanding something completely wrong?

I got an initial proxy server running and the requests e.g. `https://myserver.int/service1/` are getting correctly forwarded to `tcp://localhost:8081/` but my problem is that there are no persistent connections which is especially critical for SSL. Currently each XHR request creates a new connection even though the header `Connection: keep-alive` is set. Is this not possible or am I doing something wrong?

Here you can find a little Gist showing my proxy https://gist.github.com/nevson/25f096dd64e4571077e8

[image]

Any help would be appreciated!

cpq, 2015-07-28:
Are you trying to make something like https://github.com/cesanta/fossa/tree/master/apps/load_balancer ?

rojer, 2016-05-15:
based on your description, what you need is indeed a proxy, bu Mongoose isn't one.
load_balancer example that Sergey mentioned is close to what you need but there are currently no plans to add that functionality to mongoose.c

closed 2016-05-15 by rojer