cesanta/mongoose #564: Do not use memcmp when comparing strings

Naav, 2015-08-25:
There is a number of uses of `memcmp` to compare strings. For example (but not only), in the following extract from `mongoose.c`, lines 696-704:

    if (memcmp(str, "ssl://", 6) == 0) {
      str += 6;
      *use_ssl = 1;
    } else if (memcmp(str, "udp://", 6) == 0) {
      str += 6;
      *proto = SOCK_DGRAM;
    } else if (memcmp(str, "tcp://", 6) == 0) {
      str += 6;
    }

But in the case where `str` is shorter than the string it is being compared against, this results in an out-of-bounds access.

mmikulicic, 2015-10-30:
Hi @Naav and @nitsel88

For the record,  `memcmp` as used in this case won't cause any out of bound memory access.

`memcmp` as defined by ISO/IEC 9899:1990 states:

```
The memcmp() function returns zero if the two strings are identical, otherwise returns the difference between the first two differing bytes
```

This means that if `str` is shorter than `"ssl://"` in the above example, `memcmp` will compare the NUL terminator of `str` with a char from `"ssl://"` and then return.

Indeed ASAN doesn't issue any violations in the following code:

```c
#include <stdio.h>
#include <string.h>

int main() {
  char b1[] = "abcd";
  char b2[] = "ab";
  if (memcmp(b1, b2, 2) == 0) {
    printf("equal\n");
  }
  return 0;
}
```

Implementations of memcmp might compare full words at a time, as long as it doesn't break the mandated behaviour. Out of bound reads are fine as long as the reads don't trigger side effects and as long as tooling such as valgrind and ASAN know about it. Usually any kind of memory access control or (side effects due to memory mapped IO) happen at least at a word boundary (usually a page boundary or equivalent).

---

That said, I don't see why we shouldn't use `strncmp` here, if for anything for readability, given that we already perform by access on that string (hence the only benefit of `memcmp` 32-bit only memory access on some platform is invalidated). This particular use has already been changed.

cpq, 2015-11-21:
Thanks for reporting - as Marko said, that's not a bug.
Closing this.

closed 2015-11-21 by cpq

Naav, 2016-07-01:
Your snippet seems to be backwards with respect to the problem I described in my first post.. It should be:

```
#include <stdio.h>
#include <string.h>

int main() {
  char b1[] = "ab";
  char b2[] = "abcd";
  if (memcmp(b1, b2, 4) == 0) {
    printf("equal\n");
  }
  return 0;
}
```

And then:
```
gcc -W{all,extra,pedantic} -fsanitize=undefined,address main.c && ./a.out
=================================================================
==15064==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7fff23fe95d3 at pc 0x7f43c8b35e9f bp 0x7fff23fe95a0 sp 0x7fff23fe8d50
READ of size 4 at 0x7fff23fe95d3 thread T0
    #0 0x7f43c8b35e9e in __interceptor_memcmp ../../../../libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:414
    #1 0x4009b2 in main (/home/memcmp/a.out+0x4009b2)
    #2 0x307ec1ecdc in __libc_start_main (/lib64/libc.so.6+0x307ec1ecdc)
    #3 0x400808  (/home/memcmp/a.out+0x400808)

Address 0x7fff23fe95d3 is located in stack of thread T0 at offset 35 in frame
    #0 0x400905 in main (/home/memcmp/a.out+0x400905)

  This frame has 2 object(s):
    [32, 35) 'b1' <== Memory access at offset 35 overflows this variable
    [96, 101) 'b2'
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow ../../../../libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:414 in __interceptor_memcmp
Shadow bytes around the buggy address:
  0x1000647f5260: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f5270: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f5280: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f5290: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f52a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x1000647f52b0: 00 00 00 00 00 00 f1 f1 f1 f1[03]f4 f4 f4 f2 f2
  0x1000647f52c0: f2 f2 05 f4 f4 f4 f3 f3 f3 f3 00 00 00 00 00 00
  0x1000647f52d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f52e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f52f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x1000647f5300: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==15064==ABORTING
```

>  Out of bound reads are fine as long as the reads don't trigger side effects and as long as tooling such as valgrind and ASAN know about it.

OOB accesses, whether read or write, are undefined behaviour.

griwes, 2016-07-01:
The C standard (well, at least N1256, which looks like a reasonable draft to refer to; I don't care enough to search for a newer or older one) literally says (as normative text):

> The memcmp function compares the first n characters of the object pointed to by s1 to the first n characters of the object pointed to by s2.

In other words: both `s1[n - 1]` and `s2[n - 1]` must be valid for the call to be valid (because the normative text doesn't require that the implementation short-circuit the comparison to up-to-first-differing character; in particular, I can imagine an implementation that vectorizes or even parallelizes the comparisons so that they all happen concurrently).

Now, as for out-of-bounds accesses, the same draft says (and again, I can't imagine anything would be changed there; please do prove me wrong if you have access to any more standardized versions of the same normative text):

> Moreover, if the expression P points to the last element of an array object, the expression (P)+1 points one past the last element of the array object, and if the expression Q points one past the last element of an array object, the expression (Q)-1 points to the last element of the array object. If both the pointer operand and the result point to elements of the same array object, or one past the last element of the array object, the evaluation shall not produce an overflow; otherwise, the behavior is undefined. If the result points one past the last element of the array object, it shall not be used as the operand of a unary * operator that is evaluated.

There's two implications of the above. The first obvious one: you can't dereference a pointer that points to the element right after the end of the array (this is a trivial consequence of the last sentence).

The other consequence requires reading some more into that paragraph (it's 6.5.6.8 in the mentioned draft, by the way). `(P) + 1` is valid, because it's used for "end iterator" values (forgive me my C++ nomenclature). `((P) + 1) + 1` is *not* valid, because it's not guaranteed that it won't overflow (there's a funny additional consequence there, that is, a, say, char array can't use the last byte in virtual memory, because otherwise even `(P) + 1` would overflow...).

Hence, as I'm reading this, calling `memcmp` the way you do actually invokes UB twice:
 1. at some point it will dereference the one-past-the-end pointer (because the shorter string can potentially be identical to the beginning of the longer one!), and
 2. if `n` is bigger than the size of the shorter string by more than one, the UB happens when `memcmp` attempts to *increment the pointer*.

Please don't do this and instead fix your broken code to use `strcmp` to compare strings. Thank you.

nmmmnu, 2018-02-17:
Problem will happen if the string is in memory mapped region. Even reading outside of it will cause general protection error.

probably strcmp will be good candidate here. It also will help to omit this strange constant 6. Note strcmp does not use strlen, so performans will be basically the same.

davidhcefx, 2025-05-22:
Nice lesson on how to trigger a UB with memcmp and how to not fix a bug. Nice reading by the way!