cesanta/mongoose #565: Strange hang

erappaport, 2015-08-27:
My web server works with the version of mongoose.c available on 12/1/2014 (Windows based, multi-threaded, mostly C++ code, 32-bit). If I swap out the latest mongoose.c and mongoose.h available today, I get very strange behavior. My initial login page only comes up about 25% of the time. The rest of the time, the browser just sits there spinning. When the page comes up, I can browse around my sight, but if I stop on a page with a 10 second reload timer, once the page reloads, the browser just sits there spinning. If I exit my web server, the page shows up - so the data was sent, but the browser must think more data is coming. This happens with Firefox, Chrome, and Edge. Again - if I go back to the 12/1/2014 version of mongoose and recompile just that file (and re-link) - the entire site works properly again. This is reproducible.

Naav, 2015-08-28:
Have you tried doing a bisect to find which commit introduces this behavior?

cpq, 2015-08-28:
Thanks Ethan.
Could you run this through the Chrome's developer tools and share what network requests are, and where it is stuck please?

erappaport, 2015-08-28:
[image]
[image]
Here are the initial load and then the reload after 10 seconds.

closed 2015-10-26 by erappaport