cesanta/mongoose #571: Aborted (core dumped)

MaxSurtaev, 2015-09-22:
Initially the problem was in fossa library https://github.com/cesanta/fossa/issues/311 .
But I found the same situation repeats with mongoose too
app crashes after accepting connection

```
oracle@ub-vm:~/Downloads/mongoose-master/examples/multithreaded_restful_server$ make

cc multithreaded_restful_server.c ../../mongoose.c -o multithreaded_restful_server -W -Wall -I../.. -pthread -DNS_ENABLE_THREADS 

oracle@ub-vm:~/Downloads/mongoose-master/examples/multithreaded_restful_server$ ./multithreaded_restful_server 
Starting multi-threaded server on port 8000
epoll_ctl: Invalid argument
Aborted (core dumped)

```

I tried with gcc 4.9 on ubuntu (default install and updates) as virtual Machine
the same with gcc 4.1 on oracle linux 5 (default install and updates) as virtual Machine

Also, I tested the example simplest_web_server

added "-DNS_ENABLE_THREADS -pthread" in Makefile
added mg_enable_multithreading(nc); after nc = mg_bind(&mgr, s_http_port, ev_handler); in code
and app crushed with the same error  after accepting connection

cpq, 2015-09-25:
@MaxSurtaev can you try in again please?
Open the issue if it persists.

closed 2015-09-25 by cpq

MaxSurtaev, 2015-10-02:
the problem still there:
compiled restful_server example 

```
export SSL=openssl
oracle@ub-vm:~/Downloads/mongoose-master/examples/restful_server$ make
cc restful_server.c ../../mongoose.c -o restful_server -W -Wall -I../..  -DMG_ENABLE_SSL -lssl -lcrypto -lcrypto
oracle@ub-vm:~/Downloads/mongoose-master/examples/restful_server$ ./restful_server 
Starting RESTful server on port 8000
```

browser opens address localhost:8000 without ssl(pure HTTP)

cpq, 2015-10-04:
@MaxSurtaev could you confirm please - do you still see a problem ?

farfrontiers, 2016-01-08:
Hello! I see exactly the same problem with recently downloaded mongoose-premium-6.0.zip.

Here is the gdb backtrace:

------------------------------------------------------------------------------------------------------------
epoll_ctl: Invalid argument

Program received signal SIGABRT, Aborted.
0x00007ffff7826267 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:55
55	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7826267 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:55
#1  0x00007ffff7827eca in __GI_abort () at abort.c:89
#2  0x00000000004099d3 in mg_ev_mgr_epoll_ctl (nc=0x61d1b0, op=1) at ../../mongoose.c:2700
#3  0x0000000000409af0 in mg_ev_mgr_add_conn (nc=0x61d1b0) at ../../mongoose.c:2731
#4  0x0000000000408206 in mg_add_conn (mgr=0x7fffffffd0d0, c=0x61d1b0) at ../../mongoose.c:1766
#5  0x0000000000408c55 in mg_set_sock (nc=0x61d1b0, sock=10) at ../../mongoose.c:2062
#6  0x000000000040a608 in mg_add_sock_opt (s=0x7fffffffd0d0, sock=10, callback=0x40ab06 <forwarder_ev_handler>, opts=...) at ../../mongoose.c:3101
#7  0x000000000040a5af in mg_add_sock (s=0x7fffffffd0d0, sock=10, callback=0x40ab06 <forwarder_ev_handler>) at ../../mongoose.c:3093
#8  0x000000000040abb8 in spawn_handling_thread (nc=0x61d0e0) at ../../mongoose.c:3250
#9  0x000000000040acac in multithreaded_ev_handler (c=0x61d0e0, ev=1, p=0x7fffffffd780) at ../../mongoose.c:3277
#10 0x000000000040cbcb in http_handler (nc=0x61d0e0, ev=1, ev_data=0x7fffffffd780) at ../../mongoose.c:4020
#11 0x0000000000408359 in mg_call (nc=0x61d0e0, ev=1, ev_data=0x7fffffffd780) at ../../mongoose.c:1801
#12 0x00000000004090b4 in accept_conn (ls=0x61d010) at ../../mongoose.c:2379
#13 0x00000000004096a4 in mg_mgr_handle_connection (nc=0x61d010, fd_flags=1, now=1452292477) at ../../mongoose.c:2614
#14 0x0000000000409c56 in mg_mgr_poll (mgr=0x7fffffffdd10, timeout_ms=1000) at ../../mongoose.c:2760
#15 0x00000000004024df in main () at simplest_web_server.c:54
(gdb) frame 2
#2  0x00000000004099d3 in mg_ev_mgr_epoll_ctl (nc=0x61d1b0, op=1) at ../../mongoose.c:2700
2700	    abort();
(gdb) p epoll_fd
$1 = 0
(gdb) p *nc
$2 = {next = 0x0, prev = 0x0, listener = 0x0, mgr = 0x7fffffffd0d0, sock = 10, sa = {sa = {sa_family = 0, sa_data = '\000' <repeats 13 times>}, sin = {sin_family = 0, sin_port = 0, sin_addr = {s_addr = 0}, sin_zero = "\000\000\000\000\000\000\000"}, sin6 = {
      sa_family = 0, sa_data = '\000' <repeats 13 times>}}, recv_mbuf_limit = 18446744073709551615, recv_mbuf = {buf = 0x0, len = 0, size = 0}, send_mbuf = {buf = 0x0, len = 0, size = 0}, ssl = 0x0, ssl_ctx = 0x0, last_io_time = 1452292477, proto_handler = 0x0, 
  proto_data = 0x0, handler = 0x40ab06 <forwarder_ev_handler>, user_data = 0x0, priv_1 = 0x0, priv_2 = 0x0, mgr_data = 0x0, flags = 0}
(gdb) p *nc->mgr
$3 = {active_connections = 0x61d1b0, hexdump_file = 0x0, ctl = {0, 0}, user_data = 0x0, mgr_data = 0x0}
------------------------------------------------------------------------------------------------------------

What I see from this backtrace and debugging:

1. Mongoose server gives a supposedly unexpected epoll_fd value 0 to the epoll_ctl() call in mg_ev_mgr_epoll_ctl() function.

2. That epoll_fd goes from 'struct mg_mgr dummy', which is defined in spawn_handling_thread().

cpq, 2016-01-11:
@farfrontiers please try the HEAD version.