cesanta/mongoose #573: Remote IP missing

exmex, 2015-10-02:
There is currently no way to get the remote IP and setting an env variable because remote_ip has been removed

```
#if 0
  addenv(blk, "REMOTE_ADDR=%s", ri->remote_ip);
  addenv(blk, "REMOTE_PORT=%d", ri->remote_port);
#endif
```

and 

```
addenv(blk, "REMOTE_ADDR=%d", inet_ntoa(nc->sa.sin.sin_addr));
```

returns
```
[REMOTE_ADDR] => 0.0.0.0
```

exmex, 2015-10-04:
Also there is an issue with digest auth. If I use a global digest auth file, and request something like login.php?username=test it doesn't get through.

havonz, 2015-11-19:
spawn_handling_thread func:
add line
'c[1]->sa = nc->sa;'
before
'mg_start_thread(per_connection_thread_function, c[1]);'
to fix it

rojer, 2016-05-16:
fixed

closed 2016-05-16 by rojer