cesanta/mongoose #575: Two compiler errors: function declaration isn't a prototype and MG_DISABLE_WS_RANDOM_MASK is not defined

csaftoiu, 2015-10-05:
Compiler invocation:

    /usr/bin/cc   -pthread -std=c11 -D_GNU_SOURCE  -Dstatic_assert=_Static_assert -Wall -Wextra -Wpointer-arith -Wundef -Wvla -Wwrite-strings -Werror -Wno-error=extra -Wno-error=deprecated-declarations -Wno-error=sign-compare -Wno-error=strict-aliasing -Wno-error=type-limits -Wno-unused-parameter -Wno-error=unused-variable -Wno-error=undef -Wno-error=uninitialized -Wlogical-op -Wno-error=maybe-uninitialized -Waggregate-return -Wnested-externs -Wold-style-definition -Wstrict-prototypes -march=native -g -g3 -Og -I../../src -I../../external -Iversion -isystem /usr/local/include -MMD -MT external/CMakeFiles/mongoose.dir/mongoose/mongoose.c.o -MF "external/CMakeFiles/mongoose.dir/mongoose/mongoose.c.o.d" -o external/CMakeFiles/mongoose.dir/mongoose/mongoose.c.o   -c ../../external/mongoose/mongoose.c

Result:

    ../../external/mongoose/mongoose.c:3699:17: error: function declaration isn’t a prototype [-Werror=strict-prototypes]
     static uint32_t ws_random_mask() {
                 ^
    ../../external/mongoose/mongoose.c: In function ‘ws_random_mask’:
    ../../external/mongoose/mongoose.c:3699:17: error: old-style function definition [-Werror=old-style-definition]
    ../../external/mongoose/mongoose.c:3714:5: warning: "MG_DISABLE_WS_RANDOM_MASK" is not defined [-Wundef]
     #if MG_DISABLE_WS_RANDOM_MASK
         ^
    cc1: all warnings being treated as errors

csaftoiu, 2015-10-05:
After fixing the above (will issue a pull request):

    In file included from /usr/include/stdio.h:934:0,
                     from ../../external/mongoose/mongoose.h:101,
                     from ../../external/mongoose/mongoose.c:1:
    In function ‘fread’,
        inlined from ‘transfer_file_data’ at ../../external/mongoose/mongoose.c:3899:34:
    /usr/include/x86_64-linux-gnu/bits/stdio2.h:285:2: error: call to ‘__fread_chk_warn’ declared with attribute warning: fread called with bigger size * nmemb than length of destination buffer [-Werror]
      return __fread_chk_warn (__ptr, __bos0 (__ptr), __size, __n, __stream);
      ^
    In file included from /usr/include/x86_64-linux-gnu/sys/socket.h:251:0,
                     from /usr/include/netinet/in.h:25,
                     from /usr/include/netdb.h:28,
                     from ../../external/mongoose/mongoose.h:212,
                     from ../../external/mongoose/mongoose.c:1:
    In function ‘recv’,
        inlined from ‘mg_read_from_socket’ at ../../external/mongoose/mongoose.c:2517:23:
    /usr/include/x86_64-linux-gnu/bits/socket2.h:43:2: error: call to ‘__recv_chk_warn’ declared with attribute warning: recv called with bigger length than size of destination buffer [-Werror]
      return __recv_chk_warn (__fd, __buf, __n, __bos0 (__buf), __flags);
      ^
    ../../external/mongoose/mongoose.c: In function ‘get_mime_type’:
    ../../external/mongoose/mongoose.c:3429:22: error: function returns an aggregate [-Werror=aggregate-return]
     static struct mg_str get_mime_type(const char *path, const char *dflt,
                          ^
    ../../external/mongoose/mongoose.c: In function ‘handle_ssi_request’:
    ../../external/mongoose/mongoose.c:4354:15: error: function call has aggregate value [-Werror=aggregate-return]
         mime_type = get_mime_type(path, "text/plain", opts);
                   ^
    ../../external/mongoose/mongoose.c: In function ‘mg_send_http_file2’:
    ../../external/mongoose/mongoose.c:4452:15: error: function call has aggregate value [-Werror=aggregate-return]
         mime_type = get_mime_type(path, "text/plain", opts);
                   ^
    cc1: all warnings being treated as errors

mmikulicic, 2015-10-05:
Thanks for the report, we'd prefer zero warnings (hence the `-Werror` enforcement) as harmless warnings can obscure important ones.

Our CI is not failing. What platform (os, distro, glibc, compiler version etc) is that ?

csaftoiu, 2015-10-05:
It's on Ubuntu 12.04.2 LTS. Here's the `/usr/bin/cc -v` dump:

    $ /usr/bin/cc -v
    Using built-in specs.
    COLLECT_GCC=/usr/bin/cc
    COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.8/lto-wrapper
    Target: x86_64-linux-gnu
    Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.8.1-2ubuntu1~12.04' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
    Thread model: posix
    gcc version 4.8.1 (Ubuntu 4.8.1-2ubuntu1~12.04)

And `ldd --version`:

    $ ldd --version
    ldd (Ubuntu EGLIBC 2.15-0ubuntu10.12) 2.15
    Copyright (C) 2012 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    Written by Roland McGrath and Ulrich Drepper.

mmikulicic, 2015-10-05:
`-Waggregate-return` `-Wold-style-definition` are added by you, right?
We strive to write portable code but we don't unnecessarily limit the expressiveness of already limited ansi C. Do you know of any real platforms where those constructs might be a portability problem? If not feel free to drop those `-W` flags.

---

At first glance this seems more interesting:

```
In function ‘recv’,
    inlined from ‘mg_read_from_socket’ at ../../external/mongoose/mongoose.c:2517:23:
/usr/include/x86_64-linux-gnu/bits/socket2.h:43:2: error: call to ‘__recv_chk_warn’ declared with attribute warning: recv called with bigger length than size of destination buffer [-Werror]
  return __recv_chk_warn (__fd, __buf, __n, __bos0 (__buf), __flags);
```

I'll take a look

rojer, 2016-05-15:
we do make use of aggregate (`mg_str`) return types. we compile with `-Wall -Werror` and that's that.

closed 2016-05-15 by rojer