cesanta/mongoose #577: There is no mg_create_server

csaftoiu, 2015-10-05:
The [embedding guide](https://docs.cesanta.com/Embed.shtml) says to do:

    struct mg_server *server = mg_create_server(NULL, NULL);

Yet I did this, and get:

    error: ‘mg_create_server’ was not declared in this scope
     struct mg_server *server = mg_create_server(NULL, NULL);

I am using [this copy of mongoose.h](https://github.com/cesanta/mongoose/blob/master/mongoose.h), which does indeed [have no mg_create_server function](https://github.com/cesanta/mongoose/search?utf8=%E2%9C%93&q=mg_create_server). What am I missing?

csaftoiu, 2015-10-05:
Apparently the embedding guide is just out of date. Renaming the title of this issue.

mmikulicic, 2015-10-05:
@cpq provisionally assigning to you, since you're coordinating the doc resync.

@csaftoiu master branch is mongoose release 6.0; the doc on the main website still refers to mongoose 5.6. You can find mongoose 6.0 docs at http://docs.cesanta.com/mongoose . We are in process of cleaning up the doc. Sorry for the confusion.

csaftoiu, 2015-10-05:
Ah gotcha - thanks for the link to the latest docs!

cpq, 2015-10-06:
@csaftoiu we keep the docs for the older version of Mongoose live, like the one you mentioned, https://docs.cesanta.com/Embed.shtml.

The project page, https://github.com/cesanta/mongoose, does reference correct, up-to-date link to the documentation, the one @mmikulicic shared.

@csaftoiu could you share where did you get the https://docs.cesanta.com/Embed.shtml link from please?

csaftoiu, 2015-10-06:
@cpq: I think from the google code page, https://code.google.com/p/mongoose/ . That's where I first heard about the project - probably from googling variants on c++ embedded webserver (it's in the top 10 links). Under "Documentation", "Embedding Guide" is the first link. It leads to a "Page not found", but then I googled "mongoose Embedding Guide" and it's the first hit, so I assumed the doc was correct but the link was broken. 

It is probably a good idea to keep docs for old versions, but just make it clearer for which version it is and what the latest version is.

cpq, 2015-10-06:
@csaftoiu thank you. Unfortunately google code page is read-only, and we have no way to amend it to reflect the recent changes. So we need to live with that until google code disappears completely.

Thanks for reporting, appreciate that.

closed 2015-10-06 by cpq