cesanta/mongoose #578: core on multithreaded_restful_server example

lwmonster, 2015-10-15:
I have cloned the latest version and compiled the example multithreaded_restful_server, when running it, core occurs.

Program terminated with signal 11, Segmentation fault.
#0  0x000000000040c753 in mg_set_protocol_http_websocket ()
(gdb) bt
#0  0x000000000040c753 in mg_set_protocol_http_websocket ()
#1  0x00000000004020cb in main ()
(gdb)

Cubiverse, 2015-11-01:
Has anyone got multi-threaded working in 6.0 or later??

I moved to try to multi-threaded code directly in my application and get an abort() due to the line

Line 2682 " int epoll_fd = (intptr_t) nc->mgr->mgr_data"

Basucally mgr_data is null, and when it's used for an fd to an epoll().

Stack trace is...

mg_ev_mgr_epoll_ctl() at mongoose.c:2,695 0x4285e7	
mg_ev_mgr_add_conn() at mongoose.c:2,725 0x4286b1	
mg_add_conn() at mongoose.c:1,765 0x426f35	
mg_set_sock() at mongoose.c:2,058 0x427907	
mg_add_sock_opt() at mongoose.c:3,093 0x4291bb	
mg_add_sock() at mongoose.c:3,085 0x42915d	
spawn_handling_thread() at mongoose.c:3,240 0x4296d8	
multithreaded_ev_handler() at mongoose.c:3,267 0x4297b7	
http_handler() at mongoose.c:4,010 0x42b626	
mg_call() at mongoose.c:1,799 0x427068	
accept_conn() at mongoose.c:2,375 0x427d1c	
mg_mgr_handle_connection() at mongoose.c:2,610 0x4282ea	
mg_mgr_poll() at mongoose.c:2,754 0x428803

Cubiverse, 2015-11-01:
Can anyone confirm which version of mongoose has working multithreaded code?

losywee, 2015-11-06:
I work the multi-threaded code with the latest version.

jimax2000, 2015-11-09:
Because in latest version, MG_MGR_EV_MGR is defined to 0 by default, so mongoose will use select() and multi-threaded code works fine. But if use epoll version, it will still abort .  
... epoll_ctl: Invalid argument
now epoll_fd is 0. 

In TAG 6.0, I add two lines of code in mg_ev_mgr_epoll_ctl() in and it works, 
static void mg_ev_mgr_epoll_ctl(struct mg_connection *nc, int op) {
    int epoll_fd = (intptr_t) nc->mgr->mgr_data;
    if (epoll_fd <= 0)
        return;

However, in master branch, adding these two lines code dont solve the problem, but it is other reason.
... epoll_ctl: No such file or directory

now epoll_fd is 5, but it seems still an invalid handle.

alashkin, 2016-08-16:
Please, try latest version
epoll was deleted, a couple of mt-related bugs were fixed

closed 2016-08-16 by alashkin