cesanta/mongoose #581: Limit number of accepted connections

mmikulicic, 2015-10-21:
@lrosenth wrote:

```
I'm currently developping a IIIF-compatibe (see http:/iiif.io) image server using mongoose 6.0 (=latest). Since the conversion from JPEG2000 to TIFF, JPEG etc. may take some time, I'm running in multithreaded mode. The problem is that on heavy load to many threads are spawned, each with an open socket. So I run into the limit of open files. Of course I can increase this number, but a safer way would be to limit the number of threads. Is there a (simple) way to do this ? (as it was in version 5.4...)
```

Seems to me a reasonable feature to bring back to mongoose 6.0.

liuxyc, 2016-01-10:
this will be a useful feature, i think we need a thread pool with configurable thread number

FateStayNight, 2016-12-08:
How the progress of the thread pool？

cpq, 2016-12-08:
Mongoose is not multi-threaded. Do threadpools yourself, Mongoose is not going to help with it.
Limiting a number of connections is simply done in the `MG_EV_ACCEPT` handler, e.g.:

```
void ev_handler(...) {
   ...
  switch (ev) {
    case MG_EV_ACCEPT: {
      int num_conns = 0;
      struct mg_connection *tmp = NULL;
      while ((tmp = mg_next(c->mgr, tmp)) != NULL) num_conns++;
      if (num_conns > MAXIMUM) c->flags |= MG_F_CLOSE_IMMEDIATELY;
      ... log it ...
      break;
    }
    ...
  }
   ...
}
```

closed 2016-12-08 by cpq