cesanta/mongoose #582: Data lost between web server and browser

erappaport, 2015-10-23:
For all my installs except 1, I have no problems with the website (currently on Mongoose 6.0), but at 1 client (using Windows 2012 Server), the browser (IE and Chrome) appears not to be receiving all the data from the website. When this happens, the web server crashes about 1 in 8 times. I have reverted to Mongoose 5.6 and Mongoose 3.1 to see if they experience the same problem. The only difference is that when running Mongoose 3.1, sometimes the target browser gets a partial page to display (again, when I run the same software in my environment (Windows 2012 Server, Windows 10, and Windows 7), there are no issues).

closed 2015-10-26 by erappaport

mmikulicic, 2015-10-26:
hey @erappaport , thanks for reporting. Did you close this issue because you solved your problem?

erappaport, 2015-10-26:
Yes, this was not a mongoose issue - Ethan