cesanta/mongoose #585: publish on clibs?

zhuangya, 2015-11-10:
hey, there is a package manager for c programming language: [clib](https://github.com/clibs/clib)

any plan to publish to clib?

it would be much more convenient to use `mongoose` :smile:

cpq, 2016-04-01:
mongoose comes as an amalgamated .c/.h , just like sqlite.
So it doesn't sound feasible to use a package manager for it.

closed 2016-04-01 by cpq

mmikulicic, 2016-04-01:
@cpq, but perhaps it doesn't hurt. People who already use clib might want to just pull in mongoose right?

@zhuangya ; how hard is to make sure the clib package stays up to date?

cpq, 2016-04-01:
@mmikulicic true, but any sort of such effort, if requires our attention in the future, is a forever-ongoing tax

mmikulicic, 2016-04-01:
not sure if it's such a big tax; the only downside is that our master repo would contain an `package.json` file:

```
{
  "name": "asprintf",
  "version": "6.3",
  "repo": "cesanta.com/mongoose",
  "description": "mongoose embedded web server",
  "keywords": ["mongoose", "web", "networking", "iot", "webserver", "mqtt"],
  "src": ["mongoose.h", "mongoose.c"]
}
```

and people will install it with:

```
clib install cesanta/mongoose
```

the sources will end up in `deps/mongoose/mongoose.{c,h}`

mmikulicic, 2016-04-01:
ah, and we should have a tag matching the version number, which BTW we already have and want to keep it that way.

zhuangya, 2016-04-02:
> the only downside is that our master repo would contain an package.json file

that's true and all you have to do :)