cesanta/mongoose #588: Completion of error handling

elfring, 2015-11-11:
Would you like to add more error handling for return values from functions like the following?
- [malloc](http://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [connect_backend](https://github.com/cesanta/mongoose/blob/93beff3b65581306eac2034004563fc9a45cd752/examples/load_balancer/load_balancer.c#L264)
- [pthread_create](http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_create.html "Thread creation") ⇒ [mg_start_thread](https://github.com/cesanta/mongoose/blob/e34747d1ad014b040993eb7dac25e3395508024a/mongoose.c#L6246)

alashkin, 2016-08-24:
I'm not sure I understand rationale of this bug, but since it is a bit outdated (our fault), i'm closing it.
Feel free to close it with clarifications.

closed 2016-08-24 by alashkin

elfring, 2016-08-24:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Unchecked return values") a bit more.
Would you like to detect every error situation as early as possible?