cesanta/mongoose #589: Inability to compile Mongoose on an ESP8266 platform (RTOS)

nkolban, 2015-11-22:
Using the latest Espressif RTOS SDK I am trying to build Mongoose for execution on an ESP8266.  Having defined both RTOS_SDK=1 and MG_ESP8266=1, when I try and execute a make, I error with the following:

```
make all 
xtensa-lx106-elf-gcc -Wimplicit-function-declaration -fno-inline-functions -mlongcalls -mtext-section-literals -DRTOS_SDK=1 -DMG_ESP8266=1 -mno-serialize-volatile -I/home/kolban/git/ESP8266_RTOS_SDK/include/espressif -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip/ipv4 -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip/ipv6 -I/home/kolban/git/ESP8266_RTOS_SDK/include -I/home/kolban/git/ESP8266_RTOS_SDK/extra_include -I. -D__ETS__ -DICACHE_FLASH -DXTENSA -DUSE_US_TIMER -c mongoose.c
In file included from /home/kolban/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/dirent.h:6:0,
                 from mongoose.h:226,
                 from mongoose.c:1:
/home/kolban/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/dirent.h:10:2: error: #error "<dirent.h> not supported"
 #error "<dirent.h> not supported"
  ^
In file included from mongoose.c:1:0:
Makefile:32: recipe for target 'mongoose.o' failed
mongoose.h:228:19: fatal error: netdb.h: No such file or directory
 #include <netdb.h>
                   ^
compilation terminated.
make: *** [mongoose.o] Error 1
```

This issue requests examination of the environment to determine whether or not compilation of Mongoose for execution on an ESP8266 (RTOS) is possible and, if so, what steps or changes may be needed to achieve that goal.

rojer, 2015-11-23:
we actually do use Mongoose with ESP RTOS SDK.
try this:

-DNO_V7 -DMG_DISABLE_MQTT -DMG_DISABLE_JSON_RPC -DMG_DISABLE_COAP -DMG_DISABLE_SYNC_RESOLVER -DMG_DISABLE_HTTP_DIGEST_AUTH -DMG_DISABLE_MD5 -DMG_DISABLE_SOCKETPAIR -DMG_DISABLE_DAV -DMG_DISABLE_CGI -DMG_DISABLE_SSI -DMG_DISABLE_DIRECTORY_LISTING -DMG_MAX_HTTP_HEADERS=20 -DMG_MAX_HTTP_REQUEST_SIZE=1024 -DMG_MAX_PATH=40 -DMG_MAX_HTTP_SEND_IOBUF=1024 -DMG_ESP8266 -DMG_LOCALS  -DRTOS_SDK -DMG_LWIP

nkolban, 2015-11-23:
The repository I am trying to compile is the one found here:

https://github.com/cesanta/mongoose

When I add the definitions in your last post, my compilation output becomes:

```
make all 
xtensa-lx106-elf-gcc -Wimplicit-function-declaration -fno-inline-functions -mlongcalls -mtext-section-literals -DNO_V7 -DMG_DISABLE_MQTT -DMG_DISABLE_JSON_RPC -DMG_DISABLE_COAP -DMG_DISABLE_SYNC_RESOLVER -DMG_DISABLE_HTTP_DIGEST_AUTH -DMG_DISABLE_MD5 -DMG_DISABLE_SOCKETPAIR -DMG_DISABLE_DAV -DMG_DISABLE_CGI -DMG_DISABLE_SSI -DMG_DISABLE_DIRECTORY_LISTING -DMG_MAX_HTTP_HEADERS=20 -DMG_MAX_HTTP_REQUEST_SIZE=1024 -DMG_MAX_PATH=40 -DMG_MAX_HTTP_SEND_IOBUF=1024 -DMG_ESP8266 -DMG_LOCALS -DRTOS_SDK -DMG_LWIP -mno-serialize-volatile -I/home/kolban/git/ESP8266_RTOS_SDK/include/espressif -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip/ipv4 -I/home/kolban/git/ESP8266_RTOS_SDK/include/lwip/ipv6 -I/home/kolban/git/ESP8266_RTOS_SDK/include -I/home/kolban/git/ESP8266_RTOS_SDK/extra_include -I. -D__ETS__ -DICACHE_FLASH -DXTENSA -DUSE_US_TIMER -c mongoose.c
In file included from mongoose.c:1:0:
mongoose.h:27:23: fatal error: mg_locals.h: No such file or directory
 #include <mg_locals.h>
                       ^
Makefile:35: recipe for target 'mongoose.o' failed
compilation terminated.
make: *** [mongoose.o] Error 1
```

rojer, 2015-11-23:
ok, drop DMG_LOCALS, you don't need it

nkolban, 2015-11-23:
See attached ... it is a different set of output but still no joy.   By any chance would you have the output from a make compilation that you may have used in the past?   I am using the RTOS SDK 1.3 (the latest) from Espressif.
[buildOutput.txt](https://github.com/cesanta/mongoose/files/41254/buildOutput.txt)

rojer, 2015-11-23:
ok, i'll create an example that compiles outside our build system.

rojer, 2015-11-24:
try this: https://github.com/cesanta/mongoose/tree/master/examples/ESP8266_RTOS

rojer, 2015-11-24:
please reopen if this doesn't work for you.

nkolban, 2015-11-25:
Howdy .... I tried to build the sample and it compiled but would not link claiming that the text section was too large.  I have attached the build text file as an attachment.
[build.txt](https://github.com/cesanta/mongoose/files/43395/build.txt)

rojer, 2015-11-25:
unfortunately, linker scripts provided with the SDK are a bit (and not a small bit) wasteful. let me see what i can do.

rojer, 2015-11-25:
pushed https://github.com/cesanta/mongoose/commit/d2b9ad570fd51acce86741b73af33291ad39e644 , try that please

nkolban, 2015-11-25:
@rojer Perfect my friend.  I studied your changes and understood your recipe.  Exactly what I needed.  Not only do I now have your sample working, but I have done a quick test integrating Mongoose in my own app and it works perfectly.

closed 2015-11-25 by nkolban

rojer, 2015-11-25:
cool. glad it worked for you.

malachib, 2017-06-05:
I'm running into something very similar while trying to compile under esp-open-rtos.  Specifically I'm trying to adapt the "simplest web server ssl" example into my esp-open-rtos environment.

It seems mongoose + mbedtls needs a file system in this context, but mbedtls is (more or less) hardwired to reach to dirent.h which then results in a failure identical to the OP

I'm stuck at this point