cesanta/mongoose #590: Incorrect parameters to mg_bind sample

nkolban, 2015-11-25:
If we look at the web page found here:

`https://www.cesanta.com/developer/mongoose`

we see a usage sample called `my_app.c`.   In that sample, there is a call to `mg_bind()` that is documented as:

    mg_bind(&mgr, "1234", ev_handler, NULL); 

Unfortunately, that doesn't match the signature of `mg_bind`.  I believe the correct code should be:

    mg_bind(&mgr, "1234", ev_handler);

rojer, 2016-03-14:
doesn't seem to be the case anymore

closed 2016-03-14 by rojer

sanyaade-iot, 2018-04-13:
Hi Everybody,

I had the same problem but surprisingly work if ev_handler and NULL are swapped e.g.
nc = mg_bind(&mgr, s_http_port, NULL, ev_handler);  

For me: nc = mg_bind(&mgr, s_http_port,  ev_handler); and nc = mg_bind(&mgr, s_http_port, ev_handler, NULL); both raises compilation errors. So I have to use the above arguments.

BTW: Is there a way to disable --Werror from mos tools? 
Also can mos tool generate makefile? Or being passed a -f makefile on command line?

These features are needed by developers. 
Many thanks!

God blesses!!!

Best regards,
Sanyaade