cesanta/mongoose #593: CPU go high and never stop when ssl enabled

yzsand, 2015-12-01:
i use processhacker to make a screen capture: http://postimg.org/image/s9qbq3rpv/
windows 7,vs2010

Sample Code:

int main(void) {
	struct mg_mgr mgr;
	struct mg_connection *nc;

	mg_mgr_init(&mgr, NULL);
	nc = mg_bind(&mgr, s_http_port, ev_handler);
	
	const char *err_str = mg_set_ssl(nc,"key.pem",NULL);
	if (err_str != NULL) {
		fprintf(stderr, "Error loading SSL cert: %s\n", err_str);
		exit(1);
	}

	mg_set_protocol_http_websocket(nc);

	/* For each new connection, execute ev_handler in a separate thread */
	mg_enable_multithreading(nc);

	printf("Starting multi-threaded server on port %s\n", s_http_port);
	for (;;) {
		mg_mgr_poll(&mgr, 3000);
	}
	mg_mgr_free(&mgr);

	return 0;
}

after one connection and about 2 or 3 minutes wait, process's cpu go high and never stop

cpq, 2015-12-01:
Thanks.
Does it go high CPU witt SSL disabled ?

yzsand, 2015-12-02:
No. It works well with ssl disabled

cpq, 2015-12-02:
@rojer could you take a look at this - maybe it's reproducible on unix?

rojer, 2015-12-02:
is it 6.0 or 6.1? if yes, can you increase the debug level to `-D 4` and attach the log here?

yzsand, 2015-12-08:
Sorry for response so later!

yes. 6.0 and 6.1, both have that problem

my simple ev_handler:

static void ev_handler(struct mg_connection *c, int ev, void *p) {
	if (ev == MG_EV_HTTP_REQUEST) {
		struct http_message *hm = (struct http_message *) p;
		char reply[100];
		/* Send the reply */
		snprintf(reply, sizeof(reply), "{ \"uri\": \"%.*s\" }\n",
			(int) hm->uri.len, hm->uri.p);
		mg_printf(c, "HTTP/1.1 200 OK\r\n"
			"Content-Type: application/json\r\n"
			"Content-Length: %d\r\n"
			"\r\n"
			"%s",
			(int) strlen(reply), reply);
	}
}

after start process:

d:\testmongoose\testmongoose\mongoose.c:3312 004FFC88 using select()
d:\testmongoose\testmongoose\mongoose.c:1940 ==================================
d:\testmongoose\testmongoose\mongoose.c:1941 init mgr=004FFC88
d:\testmongoose\testmongoose\mongoose.c:3159 022F1FF8 192
d:\testmongoose\testmongoose\mongoose.c:1838 004FFC88 022F1FF8
Starting multi-threaded server on port 8000
d:\testmongoose\testmongoose\mongoose.c:3382 select @ 1449560661 num_ev=0 of 1
d:\testmongoose\testmongoose\mongoose.c:3065 022F1FF8 fd=192 fd_flags=0 nc_flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 proto ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 user ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after user flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after proto flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F1FF8 after fd=192 nc_flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3382 select @ 1449560664 num_ev=0 of 1
d:\testmongoose\testmongoose\mongoose.c:3065 022F1FF8 fd=192 fd_flags=0 nc_flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 proto ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 user ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after user flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after proto flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F1FF8 after fd=192 nc_flags=1 rmbl=0 smbl=0

after process some request, cpu go high

d:\testmongoose\testmongoose\mongoose.c:3382 select @ 1449560716 num_ev=1 of 5
d:\testmongoose\testmongoose\mongoose.c:3065 022FA148 fd=400 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022FA148 user ev=0 ev_data=004FF624 flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022FA148 after user flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022FA148 after fd=400 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022FA0A0 fd=392 fd_flags=0 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022FA0A0 proto ev=0 ev_data=004FF624 flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022FA0A0 user ev=0 ev_data=004FF624 flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022FA0A0 after user flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022FA0A0 after proto flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022FA0A0 after fd=392 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F2148 fd=212 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F2148 user ev=0 ev_data=004FF624 flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F2148 after user flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F2148 after fd=212 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F20A0 fd=200 fd_flags=1 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F9FF8 after proto flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F9FF8 after fd=368 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F20A0 after fd=200 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F1FF8 fd=192 fd_flags=0 nc_flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 proto ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 user ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after user flags=1 rmbl=0 smbl=0
[... 230 lines cut ...]
d:\testmongoose\testmongoose\mongoose.c:1892 022F2148 after user flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F2148 after fd=212 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F20A0 fd=200 fd_flags=1 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F20A0 after fd=200 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F1FF8 fd=192 fd_flags=0 nc_flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 proto ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F1FF8 user ev=0 ev_data=004FF624 flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after user flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F1FF8 after proto flags=1 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F1FF8 after fd=192 nc_flags=1 rmbl=0 smbl=0
[... cut ...]
d:\testmongoose\testmongoose\mongoose.c:1892 022FA0A0 after proto flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022FA0A0 after fd=392 nc_flags=16 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3065 022F2148 fd=212 fd_flags=0 nc_flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1866 022F2148 user ev=0 ev_data=004FF624 flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:1892 022F2148 after user flags=0 rmbl=0 smbl=0
d:\testmongoose\testmongoose\mongoose.c:3117 022F2148 after fd=212 nc_flags=0 rmbl=0 smbl=0

rojer, 2015-12-08:
thank you, that's helpful. what i see here is that select returns 1, but
none of the 5 connections have any fd_flags set when we invoke connection
handler.
i think i might've found the issue. are you able to compile from sources or
are you using binary distribution?

rojer, 2015-12-08:
ok, i'm blind. there is a connection with fd_flags = 1:

022F20A0 fd=200 fd_flags=1 nc_flags=16 rmbl=0 smbl=0

this means that there is data for it to read from the socket, and its
nc_flags indicate that it wants to read data from the socket (16 is
MG_F_WANT_READ).
but doesn't look like that takes place. the handshake has not been
completed yet (MG_F_SSL_HANDSHAKE_DONE not set) and it seems to be not
making any progress.
we're supposed to be calling mg_ssl_begin, but i don't see any debug output
from it here, which is strange.
can you please use the latest mongoose.c
<https://raw.githubusercontent.com/cesanta/mongoose/master/mongoose.c> and
mongoose.h
<https://raw.githubusercontent.com/cesanta/mongoose/master/mongoose.h> and
attach full debug output? i'd like to see entire lifetime of the connection
(here it's not clear how it was created).

yzsand, 2015-12-09:
i build from source
mongoose.c <https://raw.githubusercontent.com/cesanta/mongoose/master/mongoose.c>
mongoose.h <https://raw.githubusercontent.com/cesanta/mongoose/master/mongoose.h>

only change some output format:
mongoose.h: line 306

#include <time.h>

#define FILETIME_1970     0x019db1ded53e8000
#define HECTONANOSECONDS  10000000ui64

void GetMicroSecondFromUULTime(char* buffer)
{
	SYSTEMTIME st;
	GetSystemTime(&st);
	sprintf_s(buffer,20,"%2d:%2d:%2d.%3d ",st.wHour,st.wMinute,st.wSecond,st.wMilliseconds);
}

#define DBG(x)                              \
  if (s_cs_log_level >= LL_VERBOSE_DEBUG) {			\
	time_t rawtime;struct tm * timeinfo;			\
	char buffer [80];memset(buffer,0,80);			\
	GetMicroSecondFromUULTime(buffer);				\
	fprintf(stdout, "%-20s %8s", __func__,buffer); \
	cs_log_printf x;								\
  }

log file:

[logsslWithTime.txt](https://github.com/cesanta/mongoose/files/56469/logsslWithTime.txt)

yzsand, 2015-12-09:
and if you can supply some temporary deeper debug version to locate that strange problem, my email is: yzsand@yahoo.com
tks for your time

rojer, 2015-12-09:
actually, this latest log seems normal to me - select returns 0 evens, we
deliver EV_POLLs and repeat.
can you verify that you still get high cpu usage in this mode?

yzsand, 2015-12-10:
d:\testmongoose\testmongoose\mongoose.c:3434  2:30: 0.162 select @ 1449714600 num_ev=1 of 3

log below that line, it went into tight loop

[image]
[logsslWithTime.txt](https://github.com/cesanta/mongoose/files/57585/logsslWithTime.txt)

rojer, 2015-12-11:
i think e139bde6ae5524514fc431bea9a5b1802dea7ceb may be the solution. can you pull from master and try again?

yzsand, 2015-12-15:
@rojer @cpq 
thanks. bug fixed.
it works.

closed 2015-12-15 by yzsand