cesanta/mongoose #605: error: conflicting declaration ‘typedef struct ssl_st SSL’

MaxSurtaev, 2015-12-17:
there is a problem during static compilation with boost/asio/ssl

example of error

```
In file included from /usr/local/ssl/include/openssl/crypto.h:131,
                 from /usr/local/ssl/include/openssl/bio.h:69,
                 from /usr/local/ssl/include/openssl/conf.h:62,
                 from /usr/local/include/boost/asio/ssl/detail/openssl_types.hpp:19,
                 from /usr/local/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/local/include/boost/asio/ssl/context.hpp:27,
                 from /usr/local/include/boost/asio/ssl.hpp:19,
                 from Client.hpp:8,
                 from webserver.cpp:9:
/usr/local/ssl/include/openssl/ossl_typ.h:178: error: conflicting declaration ‘typedef struct ssl_st SSL’
/home/oracle/work/dbvisit80/dbvserver/src/libs/mongoose/mongoose.h:648: error: ‘SSL’ has a previous declaration as ‘typedef void* SSL’
/usr/local/ssl/include/openssl/ossl_typ.h:179: error: conflicting declaration ‘typedef struct ssl_ctx_st SSL_CTX’
/home/oracle/work/dbvisit80/dbvserver/src/libs/mongoose/mongoose.h:649: error: ‘SSL_CTX’ has a previous declaration as ‘typedef void* SSL_CTX’
In file included from /usr/local/include/boost/asio/ssl/context.hpp:786,
                 from /usr/local/include/boost/asio/ssl.hpp:19,
                 from Client.hpp:8,
                 from webserver.cpp:9:

```

rojer, 2015-12-18:
we typedef these structs to opaque `void *` when compiling without SSL support.
when compiling with SSL support, we use proper definitions. try compiling with `-DMG_ENABLE_SSL`.
let me know if that helps.

rojer, 2016-05-15:
closing as stale (no response)

closed 2016-05-15 by rojer