cesanta/mongoose #607: stress cgi test, mongoose dead

huojianying, 2015-12-19:
In function open_cgi_endpoint, if there is no sock left( such as in linux, you create 1024 fd and reach max limit), follow while will loop forever.

<font color=#8470FF>
  do {
    ns_socketpair(fds);
  } while (fds[0] == INVALID_SOCKET);
</font >

cpq, 2015-12-23:
I guess we shall limit spin count ?

rojer, 2015-12-23:
we should not spin at all. no socket -> 500.

cpq, 2015-12-23:
I am pretty sure I've seen spurious errors, supposedly due to interrupted syscall during socketpair creation, hence the spin.

rojer, 2015-12-23:
i see the comment, yes. but we should handle `EINTR` instead of just spinning.
i'll do it.

jmgregory, 2016-12-16:
I'm having this issue as well.  I can get around it by setting `MG_ENABLE_BROADCAST=0`, but then I also have to turn off multi-threading.

Is it possible that too many sockets are being opened but not being closed?  My system only has 2-3 concurrent clients, max.  The requests come in very fast, but they're single-threaded on the client sides.  I set `MG_DISABLE_HTTP_KEEP_ALIVE`, but it didn't seem to help.

jmgregory, 2016-12-16:
Confirmed, sockets are not being closed by mongoose.  I used `lsof` to count open socket connections while my processes were running.  My clients have about 20 total, while my server has 1013.

My event handler matches (only) on `MG_EV_HTTP_REQUEST`, and *every* code path after a match goes through the following function:

    void make_response(struct mg_connection *conn, int status_code, const char *body, bool is_json)
    {
        if (body == NULL || strlen(body) == 0) {
            mg_send_head(conn, status_code, 0, NULL);
        }
        else {
            int len = strlen(body);
            mg_send_head(conn, status_code, len, is_json ? "Content-Type: application/json" : NULL);
            mg_send(conn, body, len);
        }
        conn->flags |= MG_F_SEND_AND_CLOSE;
    }

My event handler doesn't match any other events.  Is there something I'm missing?

jmgregory, 2016-12-16:
More info: the sockets that aren't closing appear to be some kind of control sockets made by mongoose, not the HTTP request sockets.  They're created by this chunk of code at `mongoose.c:3582`:

    void mg_socket_if_init(struct mg_iface *iface) {
      (void) iface;
      DBG(("%p using select()", iface->mgr));
    #if MG_ENABLE_BROADCAST
      do {
        mg_socketpair(iface->mgr->ctl, SOCK_DGRAM);
      } while (iface->mgr->ctl[0] == INVALID_SOCKET);
    #endif
    }

I added some debug output to watch this.  For every HTTP request that comes in, mongoose makes two pairs of these DGRAM sockets.  One pair closes, the other doesn't.

jmgregory, 2016-12-16:
Think I've found the problem:  `spawn_handling_thread()` uses a dummy `mg_mgr` struct to set up connections, then transfers them to a different manager.  However, if `MG_ENABLE_BROADCAST` is set, the dummy manager creates a socket pair that is never cleaned up.  But, they can be cleaned up manually once we're `dummy`.

`mongoose.c:4731`:

       c[0] = mg_add_sock(&dummy, sp[0], forwarder_ev_handler);
       c[1] = mg_add_sock(&dummy, sp[1], nc->listener->priv_1.f);
     
    +#if MG_ENABLE_BROADCAST
    +  if (dummy.ctl[0] != INVALID_SOCKET) closesocket(dummy.ctl[0]);
    +  if (dummy.ctl[1] != INVALID_SOCKET) closesocket(dummy.ctl[1]);
    +  dummy.ctl[0] = dummy.ctl[1] = INVALID_SOCKET;
    +#endif
    +
       /* link_conns replaces priv_2, storing its value */
       poll_timeout = (intptr_t) nc->priv_2;

I'll make a pull request for this.

jmgregory, 2016-12-16:
Pull request #761 created.

rojer, 2018-01-30:
we no longer have multithreaded handler in mongoose, so this is not longer relevant.

closed 2018-01-30 by rojer