cesanta/mongoose #608: C++ example

jhindin, 2015-12-21:
I've implemented sample forward proxy on top of mongoose http library - see https://github.com/jhindin/mongoose-c--proxy.

 The proxy needs some sort of map between the servers and connections, so I've used C++ for simplicity. What is the Cesanta policy on submitting the C++ examples?

jhindin, 2015-12-21:
Missed detail: as of today, the sample proxy compiles only with changed mongoose.h files, with additional 'extern "C" {}' protection; I have not created the pull request for 'extern "C" {}' addition, in order to avoid confusion with previous pull request, 'https://github.com/cesanta/mongoose/pull/606'

cpq, 2015-12-21:
Hi Joseph,
that's great, thanks for the contribution.
Our policy is outlined at https://github.com/cesanta/mongoose#contributions
You need to fill a form at https://docs.cesanta.com/contributors_la.shtml stating that you're OK releasing your contribution under the existing Cesanta terms.

jhindin, 2015-12-21:
I am not concerned about the licensing issues; the question was about acceptability of the example implemented in C++.
I've filled the form anyway.

JohnDearlight, 2016-04-06:
FYI - the CLA form link is currently broken, at least as of 4/5/2016, 0144UTC.

https://docs.cesanta.com/contributors_la.shtml

cpq, 2016-04-08:
@JohnDearlight fixed the link, thanks !

closed 2016-04-08 by cpq