cesanta/mongoose #609: when cgi process close, we lost some data

huojianying, 2015-12-26:
As we know, in the cgi process, stdout is redirect to a local socket.

If tcp_rmem is small, when cgi process exit and the system won't handle the left data in tcp buffer, then we will lost data. 

Should we use SO_LINGER option before create cgi process?

Or  any other method to deal with it?

cpq, 2015-12-28:
Thanks. Can you describe the way to reproduce the problem, please?

huojianying, 2015-12-30:
The problem is platform-interrelated. I come across the problem on Hisi3535 embedded platform.

On that platform, this problem can be reproduced by follow method:
  1. In mongoose local endpoint don't receive data
  2. In cgi process set stdout unblocked, write until error( maybe you should write tcp_rmem + tcpwmem bytes), and exit cgi process normally.

I use tcpdump capturing packet, found when cgi process exit, mongoose receive a RST.

cpq, 2016-01-06:
Never came across Hisi3535 platform.
Is it UNIX or Windows based? Could you elaborate what CGI is used for, and in which circumstances CGI data is lost?

huojianying, 2016-01-07:
Linux. when tcp_rmem is small and cgi has lots of data to send.

rojer, 2016-05-15:
from you explanation, i understand that your process floods send and receive buffer of the stdout socket with data and exits without waiting for it to be flushed. in this case it is only normal to expect that data will be lost.
in your process, try calling `close` before exiting or even testing for stdout writeability by using select.
in either case, i don't see anything to be done on the receiving side (mongoose).

closed 2016-05-15 by rojer