cesanta/mongoose #615: Could you increae the MG_VPRINTF_BUFFER_SIZE from 100 to 200 ?

kmansoo, 2016-01-26:
Dear All,

When I ported the mongoose on eCos, it's didn't work well.
The major problem is MG_VPRINTF_BUFFER_SIZE value is too small when the mongoose use the mg_vsprintf function.

The calculated length using mg_avprintf is different between eCos and Win32.

So I think the MG_VPRINTF_BUFFER_SIZE valuss need to increase to 200 if there are no problems.

Thanks,
Mansoo

liuxyc, 2016-01-26:
looks 'mg_send' more suitable for you

rojer, 2016-05-15:
MG_VPRINTF_BUFFER_SIZE is supposed to be only for "fast path" to avoid malloc on small prints and i'd like to leave it as is to limit the stack usage. if the result is bigger than the stack buffer size, we fall back to dynamic allocation. thus, i'm more interested in this part of your report:

"The calculated length using mg_avprintf is different between eCos and Win32."

why is that, i wonder? can you elaborate?

alashkin, 2016-08-24:
I agree with @rojer 
Bug static buffers leads to memory wasting on embedded systems.

Feel free to reopen, if you still have this issue: 
>"The calculated length using mg_avprintf is different between eCos and Win32."

closed 2016-08-24 by alashkin