cesanta/mongoose #616: MG_STR

NewbProgramming, 2016-02-06:
How do I convert httpmessage -> uri to an std::string uri variable?

```
	static void EventHandler(struct mg_connection *connection, int event_id, void *event_data) {
		struct http_message *http = (struct http_message *) event_data;
		
		std::string uri("nil");

		switch(event_id) {
			case MG_EV_POLL: // ID: 0
				// Don't output any message, this gets called constantly.
			break;
			case MG_EV_ACCEPT: // ID: 1
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Accepted a connection." << std::endl;
			break;
			case MG_EV_CONNECT: // ID: 2
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Is a connection." << std::endl;
			break;
			case MG_EV_RECV: // ID: 3
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Data received." << std::endl;
			break;
			case MG_EV_SEND: // ID: 4
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Data sent." << std::endl;
			break;
			case MG_EV_CLOSE: // ID: 5
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Connection closed." << std::endl;
			break;
			case MG_EV_TIMER: // ID: 6
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Timer done." << std::endl;
			break;
			case MG_EV_HTTP_REQUEST: // ID: 100
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Is an http request." << std::endl;
				// Send custom data.
				mg_printf(connection,
							"HTTP/1.1 404 Bad Request"
					"\r\n"	"Content-Type: text/html"																		
					"\r\n"	"Connection: close"
					"\r\n"	"Cache-control: private"
					"\r\n"	"Server: lighttpd/1.4.26"
					"\r\n"	
					"\r\n"	"<!doctype html>"
							"<html>"
								"<head>"
									"<meta content=\"text/html;charset=utf-8\" http-equiv=\"Content-Type\">"
									"<meta content=\"utf-8\" http-equiv=\"encoding\">"
								"</head>"
								"<body>"
									"<h2>'%s' ERROR 404:</h2>"
									"<p>The address you typed in the address bar was not found.</p>"
									"<img src=\"picture.png\"></img>"
								"</body>"
							"</html>",
					uri.c_str()
				);
				// Send and close connection.
				connection->flags |= MG_F_SEND_AND_CLOSE;
				//
			break;
			case MG_EV_HTTP_REPLY: // ID: 101
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Is an http reply." << std::endl;
			break;
			default:
				std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
				std::cout << "Unknown." << std::endl;
			break;
		}
		return;
	}
}
```

liuxyc, 2016-02-06:
MG_STR.p is  const char *,  you know that...

NewbProgramming, 2016-02-06:
> MG_STR.p is const char *, you know that...

Yes but passing it into the std::string constructor is crashing my .exe.
Do you know what I am doing wrong? It is obviously my wrong doing.
Thank you.

mmikulicic, 2016-02-06:
The mongoose strings are not null terminated. They are string slices: a
pointer to char and a length.

mmikulicic, 2016-02-06:
IIRC, std::string has a constructor that accepts ptr and length.

NewbProgramming, 2016-02-06:
Yea so I have this:
`static void EventHandler(struct mg_connection *connection, int event_id, void *event_data) {
		struct http_message *http = (struct http_message *) event_data;
		struct mg_str *mg_uri = &http->uri;

		std::string uri(mg_uri->p, mg_uri->len);
`
Builds, but...
crashes the .exe

NewbProgramming, 2016-02-06:
Got it to work resulting code:
```
	static void EventHandler(struct mg_connection *connection, int event_id, void *event_data) {
		struct http_message *http = (struct http_message *) event_data;
		
		std::string uri("nil");
		
		if(event_id != 0) {
			std::cout << "[MONGOOSE]: Event ID (" << event_id << ") detected. ";
		}

		switch(event_id) {
			case MG_EV_POLL: // ID: 0
				// Don't output any message, this gets called constantly.
			break;
			case MG_EV_ACCEPT: // ID: 1
				std::cout << "Accepted a connection." << std::endl;
			break;
			case MG_EV_CONNECT: // ID: 2
				std::cout << "Is a connection." << std::endl;
			break;
			case MG_EV_RECV: // ID: 3
				std::cout << "Data received." << std::endl;
			break;
			case MG_EV_SEND: // ID: 4
				std::cout << "Data sent." << std::endl;
			break;
			case MG_EV_CLOSE: // ID: 5
				std::cout << "Connection closed." << std::endl;
			break;
			case MG_EV_TIMER: // ID: 6
				std::cout << "Timer done." << std::endl;
			break;
			case MG_EV_HTTP_REQUEST: // ID: 100
				std::cout << "Is an http request." << std::endl;
				
				uri.assign(http->uri.p, http->uri.len);
				// Send custom data.
				mg_printf(connection,
							"HTTP/1.1 404 Bad Request"
					"\r\n"	"Content-Type: text/html"																		
					"\r\n"	"Connection: close"
					"\r\n"	"Cache-control: private"
					"\r\n"	"Server: lighttpd/1.4.26"
					"\r\n"	
					"\r\n"	"<!doctype html>"
							"<html>"
								"<head>"
									"<meta content=\"text/html;charset=utf-8\" http-equiv=\"Content-Type\">"
									"<meta content=\"utf-8\" http-equiv=\"encoding\">"
								"</head>"
								"<body>"
									"<h2>'%s' ERROR 404:</h2>"
									"<p>The address you typed in the address bar was not found.</p>"
									"<img src=\"picture.png\"></img>"
								"</body>"
							"</html>",
					uri.c_str()
				);
				// Send and close connection.
				connection->flags |= MG_F_SEND_AND_CLOSE;
				//
			break;
			case MG_EV_HTTP_REPLY: // ID: 101
				std::cout << "Is an http reply." << std::endl;
			break;
			default:
				std::cout << "Unknown." << std::endl;
			break;
		}
		return;
	}
```

I looked at the http_client.c example.
Saw that you had to http->uri.p to access the variable.
Then I realized the crashing was because event_data was giving null information to other events being fired, so I passed the uri to the std::string while the HTTP_REQUEST was fired only.

Now it works, beautifully.
Thank you for responding and trying to help me.

closed 2016-03-14 by rojer

rhulha, 2016-09-23:
// caller needs to free memory
char * mg_str2pTEXT(struct mg_str *mgstr)
{
	char * text = (char * malloc(mgstr->len + 1);
	strncpy(text, mgstr->p, mgstr->len);
	text[mgstr->len] = '\0';
	return text;
}