cesanta/mongoose #617: HTTP Client Problem (Long polling with Redirection)

ChriD, 2016-02-06:
Hi, I have a problem where the "mg_mgr_poll" doesn't call the "mongoose_handler" callback.
It seems that there is a timing issue ore something similar. The Problem is i can't debug it because in the debugger it works fin and if a add some "sleep" methods to my code it works too

I am requesting data from a webservice which returns a HTTP redirection to another url.
The webservice is capable of longPolling and i try to do so.

The first request works and returns the Redirection. For this redirection url i am doeing a second request which works too and returns the data + the update id in the header for the long polling action

Now requesting the data of the webservice again with the long polling id in the header doesn't return anything. When i am adding some " thread sleep" between the connect and the poll it works most of the time, but not always if the request are executed to fast

Here is the code... The "sleeps"are commented.
Any idea whats going wrong? As i said i cant debug it because in debug mode it works.
Did i make some mistake?

`

    HttpRequest::HttpRequest() 
    {
        mg_mgr_init(&mongoose_mgr, this);
    }

    HttpRequest::~HttpRequest()
    { 
        mg_mgr_free(&mongoose_mgr);
    }
    
        
    void HttpRequest::runRequest(std::string _url)
    {     
        //std::this_thread::sleep_for(std::chrono::milliseconds(100));
        mg_connect_http(&mongoose_mgr, &HttpRequest::mongoose_handler, _url.c_str(), headers.c_str(), postVarsString.c_str());
        //std::this_thread::sleep_for(std::chrono::milliseconds(100));
        while (!gotResponse && !stopRequestThread)
        {
            mg_mgr_poll(&mongoose_mgr, 1000);
        }
    }
    
    // is a thread method!
    void HttpRequest::request()
    {     
        bool isRedirected;
    
        gotResponse = false;
    
        do
        {	
            isRedirected = false;
    	    runRequest(getRequestUrl());
            if (gotResponse)
            {
                 if (getResponse()->getStatusCode() == 307)
                 {                        
                     std::string redirectionUrl = getResponse()->getHeaderVar("LOCATION");
                     LUrlParser::clParseURL parsedUrl = LUrlParser::clParseURL::ParseURL(getRequestUrl());
                     setRequestUrl(parsedUrl.m_Scheme + "://" + parsedUrl.m_Host + (parsedUrl.m_Port.empty() ? "" : ":") + parsedUrl.m_Port + redirectionUrl);
                     isRedirected = true;
                  }
	    }
        }
        while(isRedirected && gotResponse)

    }

    void HttpRequest::mongoose_handler(struct mg_connection *nc, int ev, void *ev_data) 
    {
            struct http_message *hm = (struct http_message *) ev_data;
            HttpRequest *httpRequest = (HttpRequest*)nc->mgr->user_data;
            std::string response = "", responseHeader = "", responseData = "";

            if (httpRequest == nullptr)
                return;

            std::shared_ptr<HttpResponse> httpResponse;

            switch (ev) 
            { 
                case MG_EV_CONNECT:
                    if (*(int *)ev_data != 0) 
                    {
                        // ...
                       gotResponse = true;
                    }
                    break;
                case MG_EV_HTTP_REPLY:
                    nc->flags |= MG_F_CLOSE_IMMEDIATELY;
                                    
                    if (nc->recv_mbuf.len)
                    {
                       // ...
                       gotResponse = true;
                    }                                                        
                    break;
                case MG_EV_POLL:
                    break;
                default:
                    break;
            }
    }
`

cpq, 2016-02-06:
Just to clarify, is everything running in the same thread?

ChriD, 2016-02-07:
No 

Each HTTP Request creates a class wich will start a thread for the request itself.
Only the redirection will be in the same thread.

The problem that might be is that the data request (not the redirection request) will signal the main thread that he is finished and the main thread will create a new request (long polling)
After that signal the first request will destroy itself.

E.g.

    Request
        --> Create Request Object
            --> Start Request Thread
                --> doRequest
                --> Wait for result
                --> Is Redirection
                --> doRedirectionRequest
                --> Wait for result
                --> Data Here, signal main thread! SIGTHREADFINSIHED
            --> Thread finished
      --> Destroy Request Object

    SIGTHREADFINSIHED
      --> Create Request Object
         --> Start Request Thread
           --> ...

A main question: Is mongoose client able to handle 2 parallel requests  with own mg_mgr instances?

e.g.
      struct mg_mgr mongoose_mgr1;
      struct mg_mgr mongoose_mgr2;

      mg_mgr_init(&mongoose_mgr1, class1);  
      mg_mgr_init(&mongoose_mgr2, class2);
 
       ...

cpq, 2016-02-07:
Yes mongoose is able to handle many simultaneous requests.
It is an error to call IO functions from separate threads. Mongoose core is not thread-safe.

ChriD, 2016-02-07:
oh...okay.. so i have to change my logic.. Thanks.

closed 2016-02-07 by ChriD