cesanta/mongoose #619: MQTT Client not pinging - Exceeds timeout and disconnects

aaron832, 2016-02-11:
I have noticed that although mongoose.c implements a mg_mqtt_ping, it is not used.  It is also not used in any of the MQTT examples provided.  As is, ultimately after connecting to an MQTT server the connection will be lost due to keepalive.  
Noticing in code:  
`case MG_MQTT_CMD_CONNECT:
      /* TODO(mkm): parse keepalive and will */
      break;
`

mmikulicic, 2016-02-11:
yeah, thanks for "pinging" us on this TODO :-)

cpq, 2016-02-12:
Thanks @aaron832 . How urgent it is for you? Do you use mongoose for your project ?

aaron832, 2016-02-12:
Thanks.  Nope, not urgent.  I was able to work around it by sending the ping myself.

`double lastping = mg_time();
        while (1) {
                mg_mgr_poll(&mgr, 1000);
                if(mg_time() > lastping + PING_FREQ_SEC)
                {
                        mg_mqtt_ping(my_mg_connection);
                        lastping = mg_time();
                }
        }
`

aaron832, 2016-02-12:
Actually, I have noticed that sending a ping like this seems to be locking things up.  Not able to get publish messages when doing this.  Not sure why.  I have noticed also that after subscribing, the subscription complete message sometimes later after receiving a publish.

mmikulicic, 2016-02-16:
Which server are you using?

mmikulicic, 2016-03-21:
Fixed in 43709b4a86d7d0772c451f794bfdd5eb4a3a304

closed 2016-03-21 by mmikulicic

mmikulicic, 2016-03-21:
FYI: you can also schedule pings inside your event handler (just react to MG_EV_POLL events), there is no need to do it in the main poll loop.

TianyiChen, 2023-01-05:
I am seeing the same, apparently `mg_mqtt_ping` is not used and I didn't see the above commit in `master`, are MQTT clients still expected to do the ping explicitly?

cpq, 2023-01-05:
Yeah, clients may initiate pings. You might setup a timer to do it.